LORD-MicroStrain/MSCL #142: Python 3.7 or 3.8 compatibility or plans?

lamachine, 2020-12-17:
Are there plans to update the modules for these, or are they already supported?  I am using both Windows (fails on 3.7) and Raspbian Buster (still testing, but appears to fail on 3.8)

Also, the examples appear to all be for Python2, is there specific fork for 3.6 or do I just edit.  Obviously I can do a fork and shove it back to you if it is not already being done.  

  Thanks!

mglord, 2021-04-21:
Unfortunately the Python 3.6 library is not compatible with Python 3.7, 3.8 - there is a specific Python 3.6 .dll that it uses. Currently the best options are to either install and use Python 3.6 or download the source code and build MSCL locally for your target Python version (instructions here: [windows ](https://github.com/LORD-MicroStrain/MSCL/blob/master/BuildScripts/buildReadme_Windows.md)| [linux](https://github.com/LORD-MicroStrain/MSCL/blob/master/BuildScripts/buildReadme_Linux.md))

We would like to offer a 3.7 pre-built version at some point, but have no immediate plans/timeline. Same with example code - for now they just need to be updated for Python 3. If you run into any issues with building or updating the examples we would be happy to help work through them!

Sorry for the inconvenience!

lalandef, 2022-04-03:
I second this request.  Having a python 3.10 pre built library would be appreciated.

closed 2024-10-24 by github-actions[bot]