LORD-MicroStrain/MSCL #143: Event Trigger Sampling Alarm

mark5019, 2020-12-23:
Is there a function or means of being notified when the accelerometer event threshold has been triggered? I have code that sets up the connection and triggers and tells the accelerometer to start sampling, however, I would only like to start receiving the data(via the ".getData() " function) when I know that there is data being transmitted(meaning the event threshold has been exceeded).

mglord, 2021-01-05:
Once the node is in sampling mode it only 1) transmits data and 2) listens for the set to idle command. It is safe to call `BaseStation::getData()` regardless of whether or not any data has been collected or you can check `BaseStation::totalData()` which returns the number of sweeps that will be returned from `getData()` without actually returning the data. You could get misleading results from either of these if your node is transmitting diagnostic data - this will be returned along with all other DataSweeps from `getData()`. These sweeps can be differentiated because they will have a `DataSweep::samplingType()` of `DataSweep::samplingType_Diagnostic`.

Are you finding that the event trigger is not working properly and data is transmitted before the event threshold is reached?

mark5019, 2021-01-08:
Hi, Thank you for your response.

Does the the node still have to be set to idle for 'totalData()' to return the number of sweeps? Because when I call the command just before calling 'getData()' it says there are 0 sweeps yet the 'getData()' command returns data.

closed 2024-10-24 by github-actions[bot]