LORD-MicroStrain/MSCL #145: Difference in datafield

nayanagg, 2021-01-17:


mglord, 2021-01-18:
The Raw Accelerometer Vector data output is not supported on the CV5-10, but the Scaled Accelerometer Vector is.

Each of our inertial sensors supports a different subset of data outputs- there are a couple ways to check if a specific ChannelField is supported:
- MSCL: you can check if your device supports a specific output by checking if it's in the list of supported ChannelFields returned by `MipNodeFeatures::supportedChannelFields()` (`node.features().supportedChannelFields()`)
- Documentation: the list of supported outputs for the CV5-10 can be found in the [Data Communications Protocol document](https://www.microstrain.com/sites/default/files/3dm-cv5-10_dcp_manual_8500-0070.pdf) under section 3.2 Data. Data IDs can be matched to MSCL ChannelField names [here](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/MIP/MipTypes.h:MipTypes.ChannelField).

Sorry for the confusion, hope this helps!

nayanagg, 2021-01-18:
Thanks for the answer.

Also let me know the use of Accel Bias?
and how to use in python code?

mglord, 2021-03-26:
Sorry for the delayed response!

The Accel Bias data field (0x82,0x07) is an Estimation Filter output so is also not supported on the CV5-10.

In MSCL this field could be added to the Estimation Filter message format by using the `MipTypes::ChannelField::CH_FIELD_ESTFILTER_ESTIMATED_ACCEL_BIAS` enum value for supported devices.

If you'd like more information about the specifics of different output data fields and which devices support them I'd encourage you to reach out to our inertial support expert, Barry Trutor ([Barry.Trutor@parker.com](Barry.Trutor@parker.com)), for more in-depth knowledge specific to inertial devices. Please include as much information as possible about your application and the data you're trying to collect!

mglord, 2021-06-25:
I'm going to go ahead and close this, but feel free to reach out with any additional questions! Thanks!

closed 2021-06-25 by mglord