LORD-MicroStrain/MSCL #146: ¿How to export data in CSV archive?

SSI7210153, 2021-02-12:
Hi.

I'm new to programming and I work with the mscl library for python and LORD Wireless Sensors. I want to know if there is a function of the library that exporting data in CSV or if I have to make a step for step. Also, I like to know examples with the following functions, please:

- DatalogDownloader

- Histogram

¿The Histogram function makes graphics of the data?

Thank you for the help, and looking forward to your reply.

mglord, 2021-02-22:
Hi - sorry for the delayed response!

**CSV export**
We do not provide CSV export functionality within MSCL, you'll need to implement that yourself, but if you run into any issues streaming or accessing data we'd be happy to help!

**Histogram**
MSCL does not have any visualization functionality. The Histogram isn't so much a function as a specific data format that is output by our SHM-Link nodes. I suppose you could use the Histogram class to build your own histogram data over time, but we don't have any examples of this. Let me know if you're using an SHM-Link and I can provide more direction for interpreting the output histogram data.

**DatalogDownloader**
Unfortunately we don't currently have published examples for this, but I do have a code snippet in C++ that should be pretty straightforward to switch over. There is one known outstanding issue in this workflow for Python in accessing the calibration coefficients of the datalog sessions. We're working on resolving this, but in the mean time the issue and a workaround is documented in Issue 136 [here](https://github.com/LORD-MicroStrain/MSCL/issues/136#issuecomment-743303119). The workaround assumes that the calibration configuration has not changed since the time of data collection. The C++ snippet is below - let us know if you run into any issues in implementation!

```
void datalogDownload(mscl::WirelessNode& node)
{
    mscl::DatalogDownloader downloader(node);
    mscl::SampleRate currentSampleRate;
    mscl::ChannelCalMap currentCalCoefs;

    while (!downloader.complete())
    {
        // get next sweep
        mscl::LoggedDataSweep sweep = downloader.getNextData();

        // check if sample rate or cal coefficients have changed
        if (downloader.metaDataUpdated())
        {
            currentSampleRate = downloader.sampleRate();
            currentCalCoefs = downloader.calCoefficients();
        }

        // access data
        for (mscl::WirelessDataPoint data : sweep.data())
        {
            // access timestamp
            sweep.timestamp();

            // access channel name, id, number
            data.channelName();
            data.channelId();
            data.channelNumber();

            // can access actual type based on storedAs but will convert to float fine regardless
            mscl::ValueType storedAs = data.storedAs();
            float val = data.as_float();

            // if cals not already applied to sweep, apply them (optional)
            if (!sweep.calApplied())
            {
                // get cal for this data point channel
                mscl::CalCoefficients calCoef = currentCalCoefs.at(data.channelId());

                // apply coef, access unit
                val = (val * calCoef.linearEquation().slope()) + calCoef.linearEquation().offset();
                mscl::WirelessTypes::CalCoef_Unit unit = calCoef.unit();
            }
        }
    }
}
```

SSI7210153, 2021-02-26:
Thank you very much. The information is clear now.
I try to graphic data, and this is what I made. 

`# MSCL Example: BaseStationBeacon
#   This example shows how to Enable and Disable the 
#   beacon (used for Synchronized Sampling) on a base station
import time
from time import sleep
import matplotlib.pyplot as plt
import matplotlib.animation as animation
import numpy as np

# import the mscl library
import mscl
import re
import csv
import threading

# TODO: change these constants to match your setup
IP = "xxx.xxx.xxx" #WSDA-2000
PORT = xxxx
NODE_ADDRESS1 = xxxxx #G-LINK-200-OEM
NODE_ADDRESS2 = xxxxx
# flag= True
data= []
data.append([0.0])
data.append([0.0])
data.append([0.0])
data.append([0.0])
saveData={}

# Function: connectBaseStation
#   Esta función crea un objeto baseStation, realiza un ping a la base para comprobar que esté en línea y la habilita.
#   También retorna un objeto BaseStation
def connectBaseStation(IP, PORT):
     # create a Serial Connection with the specified COM Port, default baud rate of 921600
    connection = mscl.Connection.TcpIp(IP,PORT)

    # create a BaseStation with the connection
    baseStation = mscl.BaseStation(connection)

    # make sure we can ping the base station
    if not baseStation.ping():
        print("Failed to ping the Base Station")

    if baseStation.features().supportsBeaconStatus():
        status = baseStation.beaconStatus()
        print("Beacon current status: Enabled?", status.enabled(), "Time:", status.timestamp())
    
    print("Attempting to enable the beacon...")

    # enable the beacon on the Base Station using the PC time
    startTime = baseStation.enableBeacon()

    # if we got here, no exception was thrown, so enableBeacon was successful
    print("Successfully enabled the beacon on the Base Station")
    print("Beacon's initial Timestamp:", startTime)
    print("Beacon is active")

    return baseStation

# Function: disableBaseStation
#   Esta función desabilita la base
def disableBaseStation(baseStation):
    print("Wait for 3 seconds...")
    sleep(3)

    # disable the beacon on the Base Station
    baseStation.disableBeacon()

[... 256 lines cut ...]
    dataSaving.start()
    
    line_ani = animation.FuncAnimation(fig, update_line, fargs= (hl1, hl2, hl3, data), interval = 25, blit= False) #repeat= False, 
    
    plt.show()

    dataSaving.join()
    dataCollector.join()
    

except Exception as e:
    print("Error:", e)    
`
I got it, this code to print a graphic of the data and save data. But this runs very slow, any idea why? If I remove the thread and the save data function, the graph is fluid, as long as I do not remove the 30 ms wait in the graphicData function, because if I remove it, the graph freezes.

I also want to configure the node as an inclinometer. For this, I have used the function sensorOutputMode_tilt, but channels 4 and 5 are not actives, I try to use activeChannels, but so far I have not managed to use the functions well, it always gets me error:

`config = mscl.WirelessNodeConfig()
config.activeChannels.(4).enable()
`
I also tried with ChannelType, but it does not work:
`config = mscl.WirelessNodeConfig()
config.ChannelType(mscl.WirelessTypes.chType_tilt)
`
Thank you for the help, and looking forward to your reply.

SSI7210153, 2021-03-05:
I realized that the code to graph is not well appreciated, so I better attach the script.

[graficar.zip](https://github.com/LORD-MicroStrain/MSCL/files/6094019/graficar.zip)

ingeltro, 2022-05-26:
> Hi.
>
> I'm new to programming and I work with the mscl library for python and LORD Wireless Sensors. I want to know if there is a function of the library that exporting data in CSV or if I have to [...]

Hi, were you able to implement a csv conversion function? I have the same need.

closed 2024-10-24 by github-actions[bot]