LORD-MicroStrain/MSCL #147: Auto EF initialization error

san-97, 2021-02-24:
Hi,
When Auto EF initialization is enabled using MSCL library the following error is seen.
Error: The command (3410) is not supported.
setInitialFilterConfiguration is the API used for Enabling Auto EF initialization.
We are able to enable Auto EF initialization in GUI but fails using MSCL library.

Code snippet:
                cout << mscl::MSCL_VERSION.str() << endl;
                mscl::Connection connection = mscl::Connection::Serial(COM_PORT, 115200);
                mscl::InertialNode node(connection);

                cout << "Node Information: " << endl;
                cout << "Model Name: " << node.modelName() << endl;
                cout << "Model Number: " << node.modelNumber() << endl;
                cout << "Serial: " << node.serialNumber() << endl;
                cout << "Firmware: " << node.firmwareVersion().str() << endl << endl;
                bool success = node.ping();
                if(!success)
                {
                        cout << "ping Fail response:  " << success << endl;
                }
        //      node.loadFactoryDefaultSettings();
        //      node.setToIdle();
        //      node.resetFilter();
                mscl::FilterInitializationValues filterCfg;
                filterCfg=node.getInitialFilterConfiguration();
                filterCfg.autoInitialize=1;
                node.setInitialHeading((float)0);
                node.setInitialFilterConfiguration(filterCfg);
                node.setVehicleDynamicsMode(mscl::InertialTypes::AIRBORNE_HIGH_G_VEHICLE);

Application Output :
root@localhost:/home/lekha# ./lordsApp
61.0.16
Node Information:
Model Name: 3DM-GX5-45
Model Number: 6251-4220
Serial: 6251.102725
Firmware: 1.1.69
Error: The command (3410) is not supported.

Thanks

mglord, 2021-02-26:
The Initial Filter Configuration command is only supported on our latest inertial product, the GQ7, and will be used moving forward for future products with filtered data.
The GX5-45 and all our other gen 5 inertial products (GX5, CX5, CV5) use the legacy Auto Initialize Filter command which is read and written with the following functions:
```
node.getAutoInitialization();
node.setAutoInitialization(true); // false for disable auto-initialize
```

We are looking into ways to make this more intuitive in the MSCL interface and improve documentation so users don't need to rely on trial and error or refer back to the device [Data Communications Protocol](https://www.microstrain.com/sites/default/files/3dm-gx5-45_dcp_manual_8500-0064_1.pdf) for supported commands. If you have any confusion about which MSCL functions map to a specific device command in the DCP or functionality let us know and we'll be happy to help.

Many of our customers who always want the device configured the same way find it easier to configure the sensor in SensorConnect and then save the configuration as startup settings so that they don't have to worry configuring through code. 

Sorry for the confusion, let me know if this doesn't resolve the issue!

mglord, 2021-03-26:
I'm going to close this, but feel free to re-open if the `get/setAutoInitialization()` function did not resolve the issue or reach out again if you have any additional questions!

closed 2021-03-26 by mglord