LORD-MicroStrain/MSCL #150: How to interpred data in the diagnostic packet

SensicalFVO, 2021-03-25:
Hi,
using the MicroStrain G-Link-200-40G,  I am trying to interpret the data of the Diagnostic Packet. After each Diagnostic Info Interval, I get an array of 18 values. I've already checked Table 4 in the [manual](https://www.microstrain.com/sites/default/files/g-link-200_user_manual_8500-0069_0.pdf) trying to identify the meaning of each value. Unfortunately, there I count 19 rows. So I fail to correlate my data and the table.

Could you please tell me, how to interpret the 18 values received? Thanks!

mglord, 2021-03-26:
It is my understanding that the device will output only diagnostic data that is relevant to the current operating mode so the actual number of data points included in a sweep can vary (ex: when idle the device may not output sync, sample, or transmission info data). Each point has an associated channel ID that tells you what the data is - these channel IDs map to the [WirelessChannel::ChannelId enum](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/WirelessChannel.h:WirelessChannel.ChannelId) (look for entries beginning with channel_diag...).

MSCL reports the channel ID as a raw value (WirelessDataPoint::channelId()) for each data point in a sweep as well as maps it to a human-readable channel name string (WirelessDataPoint::channelName()), for example: 112 (ChannelId::channel_diag_state) maps to "diagnostic_state".

I've included a code snippet below that outputs the channel name, id, and value for each data point received to help provide context. 

Hope this helps! Let me know if anything is still unclear or I've misinterpreted your question!

```
// create connection
mscl::Connection c = mscl::Connection::Serial("COM21", 3000000);

// create base station
mscl::BaseStation bs(c);

while (true)
{
    // get collected data
    mscl::DataSweeps ds = bs.getData();

    // loop through sweeps
    for (mscl::DataSweep sweep : ds)
    {
        // check sweep type (sync, diagnostic, etc.)
        switch (sweep.samplingType())
        {
            // if diagnostic...
        case mscl::DataSweep::samplingType_Diagnostic:
        {
            cout << endl << "Diagnostic data received - Node: " << sweep.nodeAddress() << ", Data Points: " << sweep.data().size() << endl;
                    
            // loop through data points in the sweep
            for (mscl::WirelessDataPoint data : sweep.data())
            {
                cout << "    " << data.channelName() << ": " << data.as_string()
                    << endl << "        Channel ID: " << data.channelId() << endl;
            }

            break;
        }
        default:
            break;
        }
    }
}
```

SensicalFVO, 2021-03-29:
This is exactly, what I was looking for. Thank you!

closed 2021-03-31 by SensicalFVO