LORD-MicroStrain/MSCL #151: [Question] Is there a way to acquire the onBoardTemp of a 200 series node in LabVIEW?

SISKevin, 2021-03-30:
I am using an SG-Link-200-OEM and I'm trying to use the MSCL .NET in LabVIEW to determine the value of the on board temperature sensor but I do not see anything for this option in the WirelessNode invoke node.  Is there a way to acquire this temperature value?

mglord, 2021-04-07:
The sensor's internal temperature is output as part of the diagnostic packet (full diagnostic packet contents can be found in the [manual](https://www.microstrain.com/sites/default/files/sg-link-200-oem_user_manual_8500-0093.pdf) in section 4.3 Sampling Configuration, Table 5). The diagnostic packet is output at the configured interval in all sensor modes (idle, sleep, sampling). This can be configured in SensorConnect or via MSCL with the following:
```
mscl::WirelessNodeConfig config;
config.diagnosticInterval(60); // output diagnostic packet every 60 s
node.applyConfig(config); // node is a WirelessNode object
```

These packets will be collected as regular data and returned within the `DataSweeps` from `BaseStation::getData()`.

To identify a diagnostic sweep, check that the `DataSweep::samplingType()` is `samplingType_Diagnostic` ([DataSweep::SamplingType](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/DataSweep.h:DataSweep.SamplingType)).

To find the internal temperature data point within the diagnostic sweep look for the data point with `WirelessDataPoint::channelId()` equal to `channel_diag_internalTemp` ([WirelessChannel::ChannelId](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/WirelessChannel.h:WirelessChannel.ChannelId)).

If it's helpful, there is a code example for identifying diagnostic data points on issue #150 ([this comment](https://github.com/LORD-MicroStrain/MSCL/issues/150#issuecomment-808369697)).

Hope this helps! Let me know if you run into any issues!

mglord, 2021-04-07:
Just remembered, you can also get the diagnostic data on-demand with `WirelessNode::getDiagnosticInfo()` this returns a vector of `WirelessDataPoint` objects, just like `DataSweep::data()` when reading out data from `BaseStation::getData()`.

This command is only available when the node is idle.

mglord, 2021-06-25:
I hope you were able to get the info you need! I'm closing this as resolved, but let us know if you run into any additional issues!

Thanks!

closed 2021-06-25 by mglord