LORD-MicroStrain/MSCL #153: Failed to communicate with the device

chwbo, 2021-04-08:
I'm using ros_mscl for a 3DM-CV5-10 sensor with NVIDIA AGX xavier. But when I lanunched the node, I get the following error:

> nvidia@nvidia:~/catkin_ws$ roslaunch ros_mscl microstrain.launch port:=/dev/ttyUSB1
> ... logging to /home/nvidia/.ros/log/d2ac1364-9815-11eb-8541-48b02d2b8f56/roslaunch-nvidia-9745.log
> Checking log directory for disk usage. This may take a while.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://nvidia:44847/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /gx5/ros_mscl_node/baudrate: 115200
>  * /gx5/ros_mscl_node/device_setup: True
>  * /gx5/ros_mscl_node/filter_adaptive_level: 2
>  * /gx5/ros_mscl_node/filter_adaptive_time_limit_ms: 15000
>  * /gx5/ros_mscl_node/filter_angular_zupt: True
>  * /gx5/ros_mscl_node/filter_angular_zupt_topic: /moving_ang
>  * /gx5/ros_mscl_node/filter_auto_heading_alignment_selector: 1
>  * /gx5/ros_mscl_node/filter_auto_init: True
>  * /gx5/ros_mscl_node/filter_data_rate: 10
>  * /gx5/ros_mscl_node/filter_declination: 0.23
>  * /gx5/ros_mscl_node/filter_declination_source: 2
>  * /gx5/ros_mscl_node/filter_dynamics_mode: 1
>  * /gx5/ros_mscl_node/filter_enable_altimeter_aiding: False
>  * /gx5/ros_mscl_node/filter_enable_external_gps_time_update: False
>  * /gx5/ros_mscl_node/filter_enable_external_heading_aiding: False
>  * /gx5/ros_mscl_node/filter_enable_gnss_antenna_cal: False
>  * /gx5/ros_mscl_node/filter_enable_gnss_heading_aiding: True
>  * /gx5/ros_mscl_node/filter_enable_gnss_pos_vel_aiding: True
>  * /gx5/ros_mscl_node/filter_enable_magnetometer_aiding: False
>  * /gx5/ros_mscl_node/filter_enable_odometer_aiding: False
>  * /gx5/ros_mscl_node/filter_enable_vertical_gyro_constraint: False
>  * /gx5/ros_mscl_node/filter_enable_wheeled_vehicle_constraint: False
>  * /gx5/ros_mscl_node/filter_external_gps_time_topic: /external_gps_time
>  * /gx5/ros_mscl_node/filter_gnss_antenna_cal_max_offset: 0.1
>  * /gx5/ros_mscl_node/filter_heading_source: 1
>  * /gx5/ros_mscl_node/filter_init_attitude: [0.0, 0.0, 0.0]
>  * /gx5/ros_mscl_node/filter_init_condition_src: 0
>  * /gx5/ros_mscl_node/filter_init_position: [0.0, 0.0, 0.0]
>  * /gx5/ros_mscl_node/filter_init_reference_frame: 2
>  * /gx5/ros_mscl_node/filter_init_velocity: [0.0, 0.0, 0.0]
>  * /gx5/ros_mscl_node/filter_odom_child_frame_id: sensor_ned
>  * /gx5/ros_mscl_node/filter_odom_frame_id: sensor_wgs84
>  * /gx5/ros_mscl_node/filter_pps_source: 1
>  * /gx5/ros_mscl_node/filter_relative_position_frame: 2
>  * /gx5/ros_mscl_node/filter_relative_position_ref: [0, 0, 0.01]
>  * /gx5/ros_mscl_node/filter_reset_after_config: True
>  * /gx5/ros_mscl_node/filter_sensor2vehicle_frame_selector: 0
>  * /gx5/ros_mscl_node/filter_sensor2vehicle_frame_transformation_euler: [0.0, 0.0, 0.0]
>  * /gx5/ros_mscl_node/filter_sensor2vehicle_frame_transformation_matrix: [1.0, 0.0, 0.0, 0...
>  * /gx5/ros_mscl_node/filter_sensor2vehicle_frame_transformation_quaternion: [0.0, 0.0, 0.0, 1.0]
>  * /gx5/ros_mscl_node/filter_speed_lever_arm: [0.0, 0.0, 0.0]
>  * /gx5/ros_mscl_node/filter_velocity_zupt: True
>  * /gx5/ros_mscl_node/filter_velocity_zupt_topic: /moving_vel
>  * /gx5/ros_mscl_node/gnss1_antenna_offset: [0.0, -0.7, -1.0]
>  * /gx5/ros_mscl_node/gnss1_data_rate: 2
>  * /gx5/ros_mscl_node/gnss1_frame_id: gnss1_antenna_wgs84
>  * /gx5/ros_mscl_node/gnss2_antenna_offset: [0.0, 0.7, -1.0]
>  * /gx5/ros_mscl_node/gnss2_data_rate: 2
>  * /gx5/ros_mscl_node/gnss2_frame_id: gnss2_antenna_wgs84
>  * /gx5/ros_mscl_node/gpio1_behavior: 0
>  * /gx5/ros_mscl_node/gpio1_feature: 0
>  * /gx5/ros_mscl_node/gpio1_pin_mode: 0
>  * /gx5/ros_mscl_node/gpio2_behavior: 0
>  * /gx5/ros_mscl_node/gpio2_feature: 0
>  * /gx5/ros_mscl_node/gpio2_pin_mode: 0
>  * /gx5/ros_mscl_node/gpio3_behavior: 0
>  * /gx5/ros_mscl_node/gpio3_feature: 0
>  * /gx5/ros_mscl_node/gpio3_pin_mode: 0
>  * /gx5/ros_mscl_node/gpio4_behavior: 0
>  * /gx5/ros_mscl_node/gpio4_feature: 0
>  * /gx5/ros_mscl_node/gpio4_pin_mode: 0
>  * /gx5/ros_mscl_node/gpio_config: False
>  * /gx5/ros_mscl_node/gps_leap_seconds: 18
>  * /gx5/ros_mscl_node/imu_angular_cov: [0.01, 0, 0, 0, 0...
>  * /gx5/ros_mscl_node/imu_data_rate: 100
>  * /gx5/ros_mscl_node/imu_frame_id: gx5_link
>  * /gx5/ros_mscl_node/imu_linear_cov: [0.01, 0, 0, 0, 0...
>  * /gx5/ros_mscl_node/imu_orientation_cov: [0.01, 0, 0, 0, 0...
>  * /gx5/ros_mscl_node/port: /dev/ttyUSB1
>  * /gx5/ros_mscl_node/publish_filter: True
>  * /gx5/ros_mscl_node/publish_gnss1: True
>  * /gx5/ros_mscl_node/publish_gnss2: True
>  * /gx5/ros_mscl_node/publish_gps_corr: False
>  * /gx5/ros_mscl_node/publish_imu: True
>  * /gx5/ros_mscl_node/publish_relative_position: False
>  * /gx5/ros_mscl_node/raw_file_directory: /home/your_name
>  * /gx5/ros_mscl_node/raw_file_enable: False
>  * /gx5/ros_mscl_node/raw_file_include_support_data: False
>  * /gx5/ros_mscl_node/rtk_dongle_enable: False
>  * /gx5/ros_mscl_node/save_settings: False
>  * /gx5/ros_mscl_node/use_device_timestamp: False
>  * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/expected: ['ros_mscl_node: ...
>  * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/expected: ['ros_mscl_node: ...
>  * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/path: imu
>  * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/remove_prefix: ros_mscl_node
>  * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/timeout: 5.0
>  * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/imu/type: diagnostic_aggreg...
>  * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/expected: ['ros_mscl_node: ...
>  * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/path: packet
>  * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/remove_prefix: ros_mscl_node
>  * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/timeout: 5.0
>  * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/packet/type: diagnostic_aggreg...
>  * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/path: general
>  * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/expected: ['ros_mscl_node: ...
>  * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/path: port
>  * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/remove_prefix: ros_mscl_node
>  * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/timeout: 5.0
>  * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/port/type: diagnostic_aggreg...
>  * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/remove_prefix: ros_mscl_node
>  * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/timeout: 5.0
>  * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/analyzers/general/type: diagnostic_aggreg...
>  * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/path: ros_mscl_node
>  * /imu_diagnostic_aggregator/analyzers/ros_mscl_node/type: diagnostic_aggreg...
>  * /rosdistro: melodic
>  * /rosversion: 1.14.10
> 
> NODES
>   /
>     imu_diagnostic_aggregator (diagnostic_aggregator/aggregator_node)
>   /gx5/
>     ros_mscl_node (ros_mscl/ros_mscl_node)
> 
> auto-starting new master
> process[master]: started with pid [9755]
> ROS_MASTER_URI=http://192.168.2.52:11311
> 
> setting /run_id to d2ac1364-9815-11eb-8541-48b02d2b8f56
> process[rosout-1]: started with pid [9766]
> started core service [/rosout]
> process[gx5/ros_mscl_node-2]: started with pid [9773]
> process[imu_diagnostic_aggregator-3]: started with pid [9774]
> [ INFO] [1617850508.382498722]: Attempting to open serial port </dev/ttyUSB1> at <115200>
> 
> [FATAL] [1617850508.643826857]: Error: Failed to communicate with the device.
> terminate called after throwing an instance of 'mscl::Error_Communication'
>   what():  Failed to communicate with the device.
> [gx5/ros_mscl_node-2] process has died [pid 9773, exit code -6, cmd /home/nvidia/catkin_ws/devel/lib/ros_mscl/ros_mscl_node __name:=ros_mscl_node __log:=/home/nvidia/.ros/log/d2ac1364-9815-11eb-8541-48b02d2b8f56/gx5-ros_mscl_node-2.log].
> log file: /home/nvidia/.ros/log/d2ac1364-9815-11eb-8541-48b02d2b8f56/gx5-ros_mscl_node-2*.log

I tried to use it on another arm64-based device NVIDIA xavier NX, but still got the same error.   I tested it on a laptop running native Ubuntu 18.04, and  it worked well.  So, I don't know why it can't work on Xavier.

Additionally,  I tested it on Windows with Sensor Connect and MIP monitor, but it could not detect the device.

mglord, 2021-04-08:
One possible cause of communication issues like this is that the device is streaming too much data for the connection bandwidth and command responses are dropped. This could explain the failure to connect in SensorConnect and MIP Monitor, but would not explain why it consistently works on the Ubuntu laptop.

Is the device streaming data (LED blinking rapidly) or idle (LED pulsing slowly) when the ROS node is launched? If streaming can you try adding the below code to microstrain_3dm.cpp on line 300 before `//Print the device info` to set to idle immediately after establishing the connection:
```
// attempt set to idle multiple times in case responses are dropped due to data streaming
for(int i = 0; i < 3; i++)
{
  try
  {
    m_inertial_device->setToIdle();
    break; // stop trying if successful
  }
  catch(const mscl::Error& e){/*ignore*/}
}
```

Let me know if the device is not streaming or if adding the set to idle logic doesn't resolve the issue!

chwbo, 2021-04-08:
I think the device should be streaming. When I lanched the ROS node, the LED blinked rapidly twice.  After adding the above code on line 300, the LED blinked rapidly about 5 times, and also report the same error.
Now, I can use SensorConnect on windows after install a driver for CH431 serial. I follow the instructions at https://github.com/LORD-MicroStrain/ROS-MSCL/issues/41#issuecomment-800377407, but not work for me.

Thank you for your quick reply.

mglord, 2021-04-08:
Glad to hear you were able to get the device to show up in SensorConnect!

The ROS node will automatically start the device streaming on launch regardless of whether it's already streaming or not, but launching the ROS node when the device is idle allows it to connect and establish communication without worrying about connection bandwidth issues.

If you haven't already, I would encourage you to try using SensorConnect to configure the device not to stream on startup and then retry launching the ROS node while the device is idle.
In SensorConnect: Save/Load Settings -> Save as Startup Settings -> Uncheck "Stream data on startup" -> Save
The set to idle loop from before should have taken care of this, but it can't hurt to try both!

From the port name it looks like the device is connected via USB - can you confirm whether you're using a USB or serial connection? We have seen issues on some Linux environments with USB establishing the connect with anything other than a baud rate of 921600. Over USB this should not prevent communication even if the device is configured to a different baud rate.
You can launch the ROS node with a different baud rate by including the baudrate argument on launch, ex:
`roslaunch ros_mscl microstrain.launch baudrate:=921600`

Sorry for the confusion, let me know if either of these works!

chwbo, 2021-04-09:
> If you haven't already, I would encourage you to try using SensorConnect to configure the device not to stream on startup and then retry launching the ROS node while the device is idle.
> In [...]

I tried the above two methods, but still did not solve the problem.I get the same error as before.

> From the port name it looks like the device is connected via USB - can you confirm whether you're using a USB or serial connection?

Yes, I use a USB-RS232 Adapter to connect the device to xavier. run `dmesg` to show the following:
```
[ 1473.383334] usb 1-1.4.1: new full-speed USB device number 16 using tegra-xusb
[ 1473.404704] usb 1-1.4.1: New USB device found, idVendor=1a86, idProduct=7523
[ 1473.404716] usb 1-1.4.1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 1473.404722] usb 1-1.4.1: Product: USB2.0-Ser!
[ 1473.409300] ch341 1-1.4.1:1.0: ch341-uart converter detected
[ 1473.411769] usb 1-1.4.1: ch341-uart converter now attached to ttyUSB1
```

In SensorConnect, I saw the baud rate of the device was 115200. I manually clicked **set to idle**, and then tried both 115200 and 921600 baud rates when launched the ROS node in xavier, but it still didn’t work.

I tested mscl python package with same example code from [here](http://lord-microstrain.github.io/MSCL/Documentation/Getting%20Started/index.html?python#creating-an-inertialnode) and I got the following result:
```
>>> import sys
>>> sys.path.append("/usr/share/python3-mscl")
>>> import mscl
>>> connection = mscl.Connection.Serial("/dev/ttyUSB1", 115200)
>>> node = mscl.InertialNode(connection)
>>> node.ping()
False
>>> node.setToIdle()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/share/python3-mscl/mscl.py", line 6129, in setToIdle
    return _mscl.MipNode_setToIdle(self)
mscl.Error_Communication: Failed to communicate with the device.
```

chwbo, 2021-04-09:
Thank you !   I solved this problem after changing the CH341 adapter to pl2303. 
I don't know why it can't work properly with CH341 adapter. Maybe this is a problem of NVIDIA xavier CH341 driver, like [this](https://forums.developer.nvidia.com/t/troubles-with-usb-serial-adapter-ch341/165729).

mglord, 2021-04-09:
Glad you were able to get it communicating and thanks for sharing your solution!

I'm going to go ahead and close this, but feel free to re-open it or open another issue if you run into any additional problems.

closed 2021-04-09 by mglord