LORD-MicroStrain/MSCL #154: No PPS Source Options?

contagon, 2021-04-12:
We have a 3DM-GX5-45 running firmware 1.1.69.

We're trying to enable the PPS and are hitting a few roadblocks and could use some help. After successfully connecting to our IMU, we try this code:

```
// test pps stuff
mscl::PpsSourceOptions pps_options = node.features().supportedPpsSourceOptions();
printf("Getting options\n");
printf("Options found: %d\n", pps_options.size());
for(auto &s : pps_options){
     printf("Supports option %s\n", (s == 0x00) ? "PPS_DISABLED" :
                                    (s == 0x01) ? "PPS_RECEIVER_1":
                                    (s == 0x02) ? "PPS_RECIEVER_2":
                                    (s == 0x03) ? "PPS_GPIO"      :  
                                    (s == 0x04) ? "PPS_GENERATED" : "err");
  }
```

Unfortunately, we get 0 options back, and just attempting to set any of the options gives us a not supported error. Is this a bug in the MSCL library, or are we doing something wrong here?

mglord, 2021-05-21:
Sorry for the delayed response!

PPS source options are not supported by any of our gen 5 (GX5, CX5, etc.) products or earlier. Currently this is only supported on the GQ7.

You can send an external time update to the GX5-45 for time synchronization with the `InertialNode::setGpsTimeUpdate()` function. It takes a TimeFrame (week number, time of week in seconds) parameter and a value. The idea is that the week number would be updated once initially (and then again if the application is still running when it rolls over) and the time of week is updated every second.

To use this functionality, the device needs to be configured for external GNSS updates. The function to configure this is `InertialNode::setGNSS_SourceControl()` with parameter `InertialTypes::EXTERNAL_GNSS`. To then send full GNSS updates you would use the `InertialNode::sendExternalGNSSUpdate()` function.

Hope this helps!

mglord, 2021-06-25:
I hope you were able to get this working! I'm going to go ahead and close this - feel free to re-open this or create a new issue if you have any additional questions. Thanks!

closed 2021-06-25 by mglord