LORD-MicroStrain/MSCL #155: MSCL for Python 3.7?

kurtstrobel, 2021-04-21:
I’ve successfully used the MSCL library in the past with Python 2.7. I’m currently using Python 3.7 and have been unsuccessful importing the MSCL library for Python 3.6. Is there a way for me to use MSCL with Python 3.7?
 
Regards,
Kurt

mglord, 2021-04-21:
Unfortunately the Python 3.6 library is not compatible with Python 3.7 - there is a specific Python 3.6 .dll that it uses. Currently the best options are to either install and use Python 3.6 or download the source code and build MSCL locally for Python 3.7 (instructions here: [Windows ](https://github.com/LORD-MicroStrain/MSCL/blob/master/BuildScripts/buildReadme_Windows.md)| [Linux](https://github.com/LORD-MicroStrain/MSCL/blob/master/BuildScripts/buildReadme_Linux.md))

We would like to offer a 3.7 pre-built version at some point, but have no immediate plans/timeline. Sorry for the inconvenience!

mglord, 2021-05-21:
I'm closing this as answered, but if you continue to have trouble or have any additional questions feel free to re-open this or create another issue!

closed 2021-05-21 by mglord