LORD-MicroStrain/MSCL #156: Does MSCL work with old hardware?

bmsagara, 2021-04-22:
Hardware (pic attached)
Base Station: USB Base Station
Node: SG-Link (similar to the pic, but it says patent pending on it, and the pins are on the bottom)

[image]

I was trying to run the MSCL C# example using the old hardware we have (10+ years old), and I am unable to get Example 1 working (ping example).  I keep getting "Failed to ping Node 330.  Press enter to quit...".  The only thing I changed from the example was the COM_PORT and NODE_ADDRESS variable values and the MSCL_Managed.dll location.

I am able to run Node Commander and ping the nodes just fine, so I feel like the MSCL is incompatible with old hardware.

Any help would be appreciated.  Thanks!

mglord, 2021-04-23:
Unfortunately that base station is not supported by MSCL or SensorConnect - to my knowledge Node Commander is the only application we have that supports it. We could try to locate some protocol documentation if you're still interested in integrating it into your application.
It may be possible that the SG-Link can be made compatible with a firmware upgrade, but I'm unclear on the details of this - I'm getting clarification from one of our wireless product experts and will get back to you with more info on that.

If you're interested in upgrading to hardware that is compatible, I've linked the current versions of these products below, or you can get in touch with our sales team [here](https://www.microstrain.com/contact-sales) to help determine the best setup for your application.

[WSDA-200-USB](https://www.microstrain.com/wireless-sensors/WSDA-200-USB) (Base Station)
[SG-Link-200-OEM](https://www.microstrain.com/wireless-sensors/SG-Link-200-OEM)

I'll get back to you with the SG-Link-OEM upgrade details. In the meantime let me know if you would like me to try to find the USB Base Station protocol documentation or if there's any additional info we can provide!

bmsagara, 2021-04-23:
Thank you for the information!  We actually have a WSDA-200-USB base station.  I would be interested to know if a firmware upgrade on the SG-Link nodes would make them compatible with the newer base station.

I would also be interested in getting the USB Base Station protocol documentation in case a firmware upgrade does not make the nodes compatible.

Could I program the USB Base Station protocols in C#?  Or do I need to use an older language like VB6?

mglord, 2021-06-25:
Sorry for the delay!

It doesn't look like either of those devices can be made compatible with MSCL - there is a firmware upgrade available for the SG-Link that you'd be welcome to try, but I can't offer any guarantees.

If you're interested in upgrading the node please reach out to [product support](https://www.microstrain.com/contact-support) for assistance - they can help walk through how to get it upgraded.

The legacy SDK and protocol documentation can be found here: [Legacy Wireless SDK](http://updates.microstrain.com/MicroStrain-Wireless-Sensor-Networks-LXRS-SDK.zip)
The SDK has sample code in C++, VB.NET and LabVIEW but as long as you can open a connection to the device I don't see why you couldn't implement the protocol in any language you'd like.

Hope this helps!

mglord, 2021-08-23:
I'm going to go ahead and close this, but definitely let us know if you have any additional questions! Thanks!

closed 2021-08-23 by mglord