LORD-MicroStrain/MSCL #157: Inconsistent  Node Serial Numbers Returned

Rob-Hamilton, 2021-05-07:
When querying a node to get the serial number I get the full number when connecting to the WirelessNode with a base station but when I do it from the NodeDiscovery it is only the last section. I am trying to log these into an SQL Server table but the inconsistent value returned makes it hard to match up and prevent duplications.

[image]

Test Code:

    public static bool IsClosing { get; set; }
    public static mscl.BaseStation BaseStation { get; set; }
    public static mscl.Connection Connection { get; set; }
    public static mscl.WirelessNode Node { get; set; }
    public static bool IsConnected { get; set; }

    private static void Main(string[] args)
    {
        Connect("COM7", Connection.ConnectionType.connectionType_serial);
        Console.ReadKey();
        IsConnected = false;
        IsClosing = true;
    }

    private static void Connect(string address, mscl.Connection.ConnectionType type)
    {
        //10.207.142.145
        switch (type)
        {
            case Connection.ConnectionType.connectionType_serial:
                Connection = mscl.Connection.Serial(address);
                break;
            case Connection.ConnectionType.connectionType_tcp:
                Connection = mscl.Connection.TcpIp(address, 5000);
                break;
            case Connection.ConnectionType.connectionType_webSocket:                
                break;
            case Connection.ConnectionType.connectionType_unixSocket:                
                break;
        }

        BaseStation = new BaseStation(Connection);
        if (BaseStation.ping())
        {
            IsConnected = true;
            MonitorForNodeDiscoveries();
        }
    }

    private static void MonitorForNodeDiscoveries()
    {
        var tsk = Task.Run(async () =>
        {
            while (IsConnected && !IsClosing)
            {
                NodeDiscoveries discoveries = BaseStation.getNodeDiscoveries();
                if (discoveries.Count > 0)
                {
                    foreach (mscl.NodeDiscovery disc in discoveries)
                    {
                        try
                        {
                            GetNode(disc.nodeAddress());
                            await Task.Delay(100).ConfigureAwait(false);
                            Console.WriteLine($"Connected Node: {Node.nodeAddress()}\tSerial: {Node.serial()}");
                            Console.WriteLine($"Discovered Node: {disc.nodeAddress()}\tSerial: {disc.serialNumber()}");
                            break;
                        }
                        catch (Exception ex) { Console.WriteLine(ex.Message); }
                    }
                }
                await Task.Delay(500).ConfigureAwait(true);
                if (!IsConnected) { break; }
            }
        });
    }

    private static void GetNode(uint nodeAddress)
    {
        try { Node = new mscl.WirelessNode(nodeAddress, BaseStation); }
        catch (Exception) { Node = null; }
    }

mglord, 2021-05-07:
The full serial, what `WirelessNode::serial()` returns, is a combination of the model number and serial number, which is intended. For some reason (probably just an oversight) that combined string was not made available from the `NodeDiscovery` object.

I have added a task to make the combined formatted string available from the `NodeDiscovery` object, but in the meantime, you might be able to use one of the following as a workaround:
* Use the node address, available from both the `NodeDiscovery` and `WirelessNode` objects, as the unique identifier. I will follow up with someone on our Wireless team to ask why nodes have two unique identifiers (address and serial).
* Build the full serial from the `NodeDiscovery`'s model and serial numbers. The code used by the `WirelessNode` to do this is:
```
std::string formatSerialStr(mscl::WirelessModels::NodeModel model, uint32_t serial)
{
  //split the model into its 2 pieces
  uint16 model = static_cast<uint16>(model / 10000);
  uint16 modelClass = model % 10000;

  //build the string result
  std::stringstream modelStr;
  modelStr << std::setfill('0') << std::setw(4) << model;

  std::stringstream modelClassStr;
  modelClassStr << std::setfill('0') << std::setw(4) << modelClass;

  std::stringstream serialStr;
  serialStr << std::setfill('0') << std::setw(5) << serial;

  return modelStr.str() + "-" + modelClassStr.str() + "-" + serialStr.str();
}
```

Sorry for the inconsistency - let me know if you run into issues with these workarounds!

Rob-Hamilton, 2021-05-10:
@mglord 

Thank you for the quick response. I am not very fluent with C++ but I think I have it converted over to C#, at least it gives me the same answer. Please check this and see if you see any issue with my conversion.

[image]

    private static string GetDiscoveredNodeSerial(NodeDiscovery disc)
    {
        var discModel = disc.model();
        var uIntModel = ((uint)discModel);
        var model = uIntModel / 10000;
        var modelClass = uIntModel % 10000;
        var serial = disc.serialNumber();

        return $"{model}-{modelClass}-{serial}";
    }

mglord, 2021-05-10:
Looks good to me!

And to clarify regarding node address vs serial number for identification: the serial number is a true unique identifier across all our products but the node address only needs to be unique within a single network. The node address is a much smaller number (16-bit instead of 32) and allows packets to specify a source/destination within a network without including the full 32-bit serial number.