LORD-MicroStrain/MSCL #159: Do you have an  idea that we create the software with C++10?

yuukireo, 2021-06-01:
We use the LORD wireless sensing device, and we will create the software by my self.
Also, we will write it with the C++10.
However, The LORD software enginior said it is written in C++11, 
and C++ 11-specific objects in the interface that would prevent compatibility with older versions.

We will try to create the software but it is not possible to include of Mutix with C++10, 
Do you have an idea that you can repair the MSCL to be able to use C++10?

I know it is hard to say to you but…
If you have any required condition to revise software as customer's request, please let me know anything freely.

We are lookimg for hearling from you soon.
Thank you for your big support.

Best Regards,

yuukireo, 2021-06-03:
Thank you for your suppor.
I can make it. We will try to build it.

yuukireo, 2021-06-04:
In the previous questions, I understand that the latest version of MSCL is not compatible with VisuaStudio2010 
and previous versions of C ++.
However, it is difficult to replace with the latest version because the customer's system is configured with VS2010.
Therefore, customers are strongly requested to customize MSCL so that it can be used in C ++ 10.
Can you develop for C ++ 10?
Also, if you have any conditions (cost / period, etc.) for development / examination, please let us know.

Thnak you.
Best,

mglord, 2021-06-08:
I'm sorry, we're just not able to take on making MSCL compatible with Visual Studio 2010. One of the reasons we made MSCL open-source is for maximum flexibility - you are welcome to download the source and make whatever changes necessary to work within your development environment.

As mentioned above, however, depending on the scope of functionality you need it will likely be easier to build your own interface for the subset of commands and data parsing you need. If you let us know your setup and intended use case we'd be happy to help determine the best workflow and which commands and data you will need.

I would be happy to help answer any additional functional or technical questions, but I would encourage you to direct any more business-related questions to Traci Stevens ([Traci.Stevens@parker.com](Traci.Stevens@parker.com)). Thanks!

mglord, 2021-06-25:
I'm going to go ahead and close this, but please feel free to reach out to further discuss implementation options and required functionality for your application!

Thanks!

closed 2021-06-25 by mglord