LORD-MicroStrain/MSCL #160: boost and Open SSL for 32bit.

yuukireo, 2021-06-02:
We are not sure which file is the good choice for boost and openSSL with operating 32bit.

Which folder is the best choice with operating 32bit ? 
Also, which lib file, and header file are the best choice for it??
We have to link the best header file and lib file, but we couldn't find it.
We want to know which is the best choice clealy.

Colud you give us some advice.

Thank you for your advice.
Yuki

mglord, 2021-06-02:
I'm assuming you're looking for the builds compatible with Visual Studio 2010 - please correct me if I'm mistaken!

You likely will not need OpenSSL, but they do have a build of v1.1.0 for VS2010 [here](https://www.npcglib.org/~stathis/blog/precompiled-openssl/) listed for the MSVC 2010 compiler (openssl-1.1.0f-vs2010.7z). This should include the debug and release builds for both 32- and 64-bit systems. For 32-bit you would just include and link to the "include" and "lib" directories instead of "include64" and "lib64".

Boost specifies differentiates their builds based on the MSVC compiler version your using - Visual Studio 2010 uses MSVC 10.0 (version chart [here](https://en.wikipedia.org/wiki/Microsoft_Visual_C%2B%2B#Internal_version_numbering)). Boost has a binary for the MSVC 10.0 compiler on a 32-bit system [here](https://sourceforge.net/projects/boost/files/boost-binaries/1.68.0/) (boost_1_68_0-msvc-10.0-32.exe). Once installed you would include and link to the following:
- Include directory: <boostInstallPath>/boost 
- Link directory: <boostInstallPath>/lib32_msvc-10.0 
- Link to: libboost_system-vc100-mt-s-x32-1_68.lib

As mentioned in issue #159, you will likely run into compatibility issues using MSCL with Visual Studio 2010, but I see no reason you wouldn't be able to use Boost, OpenSSL in your own implementation.

Hope this helps!

yuukireo, 2021-06-03:
Thank you for your suppor.
There is one more things I’ll ask you about.

Where is the "ibboost_system-vc140-mt-sgd-x32-1_68.lib"??
I couldn't find it.......

Could you give us an information where it is cleary?

Thank you for your support.

Best,

yuukireo, 2021-06-03:
I'm so sorry......
We will use OpenSSL, and Boost for Visual studio2017.

Do you have an idea about it?
Thank you for your support.

mglord, 2021-06-09:
Sorry for the delay!

Once Boost is installed, the `libboost_system-vc100-mt-s-x32-1_68.lib` library file (Visual Studio 2010, 32-bit version of the file) can be found in the link directory `<install location>/lib32_msvc-10.0`.

**For the Visual Studio 2017 version:** anywhere in the directory and files names that has `vc100`, `msvc-10.0` for the Visual Studio 2010 version will be `vc141` and `mscv-14.1` for the Visual Studio 2017 version:
- Include directory: /boost
- Link directory: /lib32_msvc-14.1
- Link to: libboost_system-vc141-mt-s-x32-1_68.lib

Hope this helps - thanks for your patience!

mglord, 2021-06-25:
I hope you were able to find and link the files to get this working! I'm going to go ahead and close this, but please feel free to re-open this or create another issue if you have further questions or run into any additional problems!

Thanks!

closed 2021-06-25 by mglord