LORD-MicroStrain/MSCL #161: [Data Doenload Session using MSCL]

YukiMiwa30, 2021-06-09:
I want to get the log data of G-LINK-200 using MSCL.
I would like to implement SensorConnect "Download Datalog Sessions" using MSCL. 
What classes and functions do I use?

mglord, 2021-06-09:
Once you've established communication with a node, all you should need from there is the [DatalogDownloader](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/DatalogDownloader.h:DatalogDownloader) class. This class handles retrieving datalog session information and keeping track of the download progress.

We don't currently have published examples for this, but below is a quick C++ snippet that should pretty much cover it. Let us know if you run into any questions or issues during implementation!

```
void datalogDownload(mscl::WirelessNode& node)
{
    mscl::DatalogDownloader downloader(node);
    mscl::SampleRate currentSampleRate;
    mscl::ChannelCalMap currentCalCoefs;

    while (!downloader.complete())
    {
        // get next sweep
        mscl::LoggedDataSweep sweep = downloader.getNextData();

        // check if sample rate or cal coefficients have changed
        if (downloader.metaDataUpdated())
        {
            currentSampleRate = downloader.sampleRate();
            currentCalCoefs = downloader.calCoefficients();
        }

        // access data
        for (mscl::WirelessDataPoint data : sweep.data())
        {
            // access timestamp
            sweep.timestamp();

            // access channel name, id, number
            data.channelName();
            data.channelId();
            data.channelNumber();

            // can access actual type based on storedAs but will convert to float fine regardless
            mscl::ValueType storedAs = data.storedAs();
            float val = data.as_float();

            // if cals not already applied to sweep, apply them (optional)
            if (!sweep.calApplied())
            {
                // get cal for this data point channel
                mscl::CalCoefficients calCoef = currentCalCoefs.at(data.channelId());

                // apply coef, access unit
                val = (val * calCoef.linearEquation().slope()) + calCoef.linearEquation().offset();
                mscl::WirelessTypes::CalCoef_Unit unit = calCoef.unit();
            }
        }
    }
}
```

YukiMiwa30, 2021-06-25:
Thank you for your reply.

When I run the source code that I've told you, I get an error.
The error occurs after hundreds of iterations, but not always with the same number of iterations.

By adding a 50ms wait process, it took longer for the error to occur, but the error still occurs.
Please tell me the cause and countermeasures.

[Error details]
Failed to download data from the Node.

mglord, 2021-06-25:
This is likely just a result of random communication failures that happen occasionally with wireless communication. The best way around this is to catch the exception on `downloader.getNextData()` and retry it. In order to ensure downloads don't fail unless there's a legitimate issue, SensorConnect allows for up to 100 retries each time `getNextData()` is called. In most cases more than a couple retries should not be needed, but you can adjust the number to what seems reasonable and works for your application.

Below is the while loop from the above example but with similar retry logic that we use in SensorConnect added - also note the metadata check and looping through the data points has been removed to make the example less cluttered. The same code can be added back in the `try` block where indicated. Hope this helps!
```
const uint8_t MAX_RETRIES = 100;
uint8_t retryCounter = MAX_RETRIES;

while (!downloader.complete())
{
    try
    {
        // get next sweep
        mscl::LoggedDataSweep sweep = downloader.getNextData();
        
        // successfully read data, reset retry counter
        retryCounter = MAX_RETRIES;
        
        // check metadata and access data here (see example in previous comment)
    }
    // No Data exception indicates an actual failure response - don't retry
    catch (mscl::Error_NoData)
    {
        throw;
    }
    // if you want to only catch specific error types mscl::Error_Communication
    // and mscl::Error_NodeCommunication should account for most random communication failures
    catch (...)
    {
        // if attempted max retries don't retry again
        if(retyCounter <= 0)
        {
            throw;
        }
        
        // decrement counter and retry
        retryCounter--;
    }
}
```

b-mach, 2021-06-27:
hi mglord
Do you have the above code available in csharp? I want to do this but have all my code in c-sharp.

mglord, 2021-06-28:
Unfortunately we don't currently have this example published in multiple languages, but the above are in C++ and should be pretty straightforward to convert. All class names and function calls are the same.

Syntax is similar - the following changes should just about cover it:

- Replace accessor `::` with `.` ie `mscl::DatalogDownloader` turns into `mscl.DatalogDownloader`
- Replace `uint8_t` (or any uint types) with `byte` (8-bit), `ushort` (16-bit) or the [C# type of the same size](https://docs.microsoft.com/en-us/dotnet/csharp/language-reference/builtin-types/integral-numeric-types#characteristics-of-the-integral-types)
- `void datalogDownload(mscl::WirelessNode& node)` - remove `&` from `WirelessNode&`. If I'm remembering correctly you don't need `ref` because it passes a copy of the reference and you're not changing the value of the `node` parameter.

Hope this helps, let us know if you run into any issues!

b-mach, 2021-06-29:
Thanks. Just also needed to initialize by changing the line to
mscl.DatalogDownloader downloader = new mscl.DatalogDownloader(node);

mglord, 2021-06-29:
Good catch, thanks!

mglord, 2021-08-23:
I'm going to go ahead and close this, but if you have any additional questions definitely feel free to re-open this or create another issue! Thanks!

closed 2021-08-23 by mglord