LORD-MicroStrain/MSCL #162: How to perform a Shunt Calibration using MSCL: LabVIEW?

aerophd2001, 2021-06-10:
Using MSCL in LabVIEW, how would one go about performing a shunt calibration?

My current hardware: WSDA-200-USB, SG-Link-200-OEM.

mglord, 2021-06-11:
The easiest option for performing a shunt cal is to go through the Configure->Calibration menu in SensorConnect, but it's certainly available in MSCL as well if your application requires it.

Provided you have all the required info, executing the shunt cal in MSCL is a single function call in MSCL (`WirelessNode::autoShuntCal()`) to run the shunt cal and then the returned slope and offset need to be applied to the node. It takes two parameters, the channel mask of the channel you want to perform the operation on and the shunt cal info - if you don't need any additional info about the command parameters, you can jump right to the **Run the Shunt Calibration and save results** section below. Some of the shunt cal info is current or desired device settings, so if you don't already have that information you will need to read it from the device. We don't have a LabVIEW example, but hopefully the below information helps make implementation fairly straightforward.

Apologies for the essay, I just wanted to include as much info as possible as I was informed this may be time-sensitive.

Let us know if anything needs clarification or you run into any issues during implementation!

### Channel Mask
The for this particular operation the channel mask will only represent a single channel, but because the same structure can be used to represent multiple channels in a single value the mask value is a bitfield where a channel is represented by turning on the bit at the index of the channel number. For example:
- Channel 1 - mask bitfield: **0001**, mask value: **1**
- Channel 2 - mask bitfield: **0010**, mask value: **2**
- Channel 3 - mask bitfield: **0100**, mask value: **4**

### Shunt Cal Info
The other parameter is a `ShuntCalCmdInfo` object. This contains all other required variables:
- `useInternalShunt`: true for internal, false for external
- `numActiveGauges`: number of active gauges or set to 1 if you'd rather specify bridge factor
- `gaugeFactor`: gauge factor - if specifying bridge factor, this value should be `bridgeFactor * gaugeFactor`
- `gaugeResistance`
- `shuntResistance`
- `inputRange`: device configuration - see below
- `hardwareOffset`: not supported by SG-Link-200, set to 0
- `excitationVoltage`: device configuration - see below

### Device configuration parameters
There are a few parameters that are other configured values on the device. These are parameterized instead of using the currently configured values so that the shunt cal can be run with a pending configuration. If you have desired input range, hardware offset, and excitation voltage values that are pending or you know are configured to the device. To read these values:

- `inputRange`: all the channels that support shunt cal on the SG-Link-200 share an input range so the channel mask to read the setting is bitfield: **0111**, value: **7**:
```
mscl::ChannelMask chs;
chs.fromMask(7);
node.getInputRange(chs);
```
- `excitationVoltage`: this setting applies to all channels: `node.getExcitationVoltage()`

### Run the Shunt Calibration and save results
Once you have all the command parameter values it's pretty straightforward. Keep in mind that the `autoShuntCal()` function may take a couple seconds to return the results. The function does not resolve until it receives the results (or times out).
```
// build channel mask  - do this for each channel
mscl::ChannelMask chMask;
chMask.enable(3); // represent channel 3 (mask value will be 4 (0100))

mscl::ShuntCalCmdInfo shuntCalInfo;
// set parameters here

// run auto shunt cal operation on mscl::WirelessNode object (node)
mscl::AutoShuntCalResult result = node.autoShuntCal(chMask, shuntCalInfo);

// build WirelessNodeConfig object to set up configuration to apply
mscl::WirelessNodeConfig config;

// configure slope and offset for channel based on shunt cal results
config.linearEquation(chMask, mscl::LinearEquation(result.slope(), result.offset()));

// apply configuration to node
node.applyConfig(config);
```

aerophd2001, 2021-06-11:
Is there no function to simply engage/disengage the shunt resistor? I can do the math myself for calculating the calibration. I know that with the previous generation of your hardware, there was a bit you could toggle within the eeprom that would do this.

mglord, 2021-06-11:
Sorry, I'm not sure about that - I'll have to run this by one of our wireless engineers and get back to you!

Are you able to provide a little more detail about what specific functionality you're trying to implement? If it's helpful, the `autoShuntCal()` function also returns the following information in the `AutoShuntCalResult` object for the data sampled during the shunt cal operation:

- `baseMedian`: the median of the baseline data
- `baseMin`: the minimum of the baseline data
- `baseMax`: the maximum of the baseline data
- `shuntMedian`: the median of the shunted data
- `shuntMin`: the minimum of the shunted data
- `shuntMax`: the maximum of the shunted data

aerophd2001, 2021-06-11:
I've just found the autoShuntCal() function and I'm in the process of coding it up to test.

Speaking purely as an engineer, it would be useful to have the ability to toggle the shunt resistor on and off. As I mentioned before, I can do the math myself for computing the effect of the shunt resistor on the strain calibration. All I need is to measure, toggle the shunt on, measure again, and toggle the shunt back off. From there I can do the rest.

aerophd2001, 2021-06-17:
I have successfully implemented the autoShuntCal() function. Thanks for the help!

mglord, 2021-06-25:
I'm glad you were able to get the shunt cal function working! I'm going to go ahead and close this, but wanted to note that one of our wireless engineers clarified that the ability to manually engage/disengage the shunt resistor was added to the V-Link-200 in firmware version 12.49145. Currently this is only available for the V-Link-200.

This functionality is controlled by setting EEPROM 490:

- Set to 0 for normal node operation (shunt resistor not engaged) (default)
- **While sampling**: when set to any value 1 or higher the node will cycle the shunt resistor to each of the differential channels one at a time for one second each
- **While idle**: when set to a specific channel number (only valid for differential channels) the shunt resistor will be applied to the specified channel during a poll node request. By polling the node both with and without EEPROM 490 set a calibration can be done based on the data output.

This is not yet available for the SG-Link-200, but I saw in the other thread that you also have a V-Link-200 and wanted to document this in case it's helpful to you or anyone else.

Let us know if you have any questions about implementation or have any further questions regarding the `autoShuntCal()` function!

closed 2021-06-25 by mglord