LORD-MicroStrain/MSCL #163: Serial number from listBaseStations() does not match actual serial number on base station.

Rob-Hamilton, 2021-06-11:
LORD gurus,
When I get the serial number from mscl.Devices.listBaseStations() via the returned mscl.DeviceInfo it does not actually match what is on the base station nor what I get back after connecting to a base station and getting the BaseStation.serial(). The BaseStation.serial() correctly returns what is actually on the device. It seems to me that something as uniquely identifying as a serial number should always be the same and should match the physical serial number on the device. Also when I use the mscl.WsdaFinder() there isn't even a property for getting the serial number at all. I would think this would be a desired attribute to know when you are discovering base stations, I know I need it.

### Example:  WSDA-200-USB
listBaseStations()    ==> 388221071ADDE71195B01B45CB70D25B
BaseStation.serial() ==> 6307-2040-91766

### Example:  WSDA-2000
listBaseStations()    ==> W020000000090355
BaseStation.serial() ==> 6314-2000-90355

### listBaseStations() method
```
  foreach (KeyValuePair<string, mscl.DeviceInfo> item in mscl.Devices.listBaseStations())
  {
       var serial = item.Value.serial();
  }
```
###  BaseStation.serial() method

```
  BaseStation = new mscl.BaseStation(Connection);
  var serial = BaseStation.serial();       
```

mglord, 2021-06-15:
Looked into this and found the same behavior. I don't see the serial stored in the device EEPROM anywhere in the information exposed by the driver - it may not be accessible without opening the connection and asking the device.

I'll have to run this by one of our wireless engineers to find out a) if the device serial/model number is available without opening the connection to the device and b) what the `listBaseStations()` serial value is.

A workaround would be to use the port name (`item.Key`) and baud rate (`item.Value.baudRate()`) to create a `Connection` and `BaseStation` object to check the serial for each entry returned from `listBaseStations()`.

Looking at the `WsdaFinder` source, the `WsdaInfo::name()` property returns a private variable named `m_serial`. It looks like this value will be in the same format of the `listBaseStations()` entry serial (W020000000090355). For WSDA-2000s, both using `listBaseStations` and the `WsdaFinder` the last five or six digits will be the same as the "serial" section of the device-reported serial number. As far as I'm aware, there is only one model and option number for the WSDA-2000 (will confirm with wireless team) so you should be able to safely replace the "W02000000..." with "6314-2000-".

Sorry for the confusion around this - I'll reach out to our wireless team and see if I can track down better explanations for this behavior and how we may be able to improve it.

Rob-Hamilton, 2021-06-17:
@mglord 
Thank you for looking into this. Unfortunately I don't think that splicing a constant with the returned value would be very future proof. One day it may change and then we are back to figuring out the new prefix and modifying all of my code, same with new models that would come out. I think the only real solution would be to get the actual serial number when the base station is discovered. Until then I will just have to record both to make sure I can cross reference them.

Again thank you for all of your support on this and the many other issues I seem to have :-)

closed 2024-10-24 by github-actions[bot]