LORD-MicroStrain/MSCL #164: Inertial Node fails to establish a serial connection

bharatm11, 2021-06-14:
I am using a GX5-15 VRU. It fails to establish a serial connection with the unit and throws out a communication error exception with the message "Failed to communicate with the device".

My C++ code which was taken from the [Quick Start Guide](http://lord-microstrain.github.io/MSCL/Documentation/Getting%20Started/index.html#inertial) used to work on a different build environment. This issue started popping up when we switched environments. 

Does MSCL support the following configuration?
1. Windows 10 Enterprise LTSC Version 1809 OS Build 17763.1457
2. MSVC 2017 15.9 compiler?

The VRU is able to communicate in the new development environment using python, MATLAB, and MIP Monitor.

mglord, 2021-06-15:
The two most common things that cause the communication failure error are 1) incorrect baud rate and 2) bandwidth issues due to streaming. I'm not sure that's what's happening here due to the success communicating on other environments and in different applications on the same environment, but worth a check - I've included info and troubleshooting steps about each of these below.

I don't know of any specific reason MSCL shouldn't work with and LTSC version of Windows, especially since you are able to communicate with the device via MIP Monitor. If you're able to install [SensorConnect](https://www.microstrain.com/software/sensorconnect) I'd be interested to see if device communication is successful there. SensorConnect is built on top of MSCL so if it works MSCL should too.

Another thing you could try (if not already) is installing the version of Boost specific to the MSVC 2017 15.9 compiler - based on [this chart](https://en.wikipedia.org/wiki/Microsoft_Visual_C%2B%2B#Internal_version_numbering) that compiler uses MSVC++ version 14.16. The corresponding Boost installer can be found [here](https://sourceforge.net/projects/boost/files/boost-binaries/1.68.0/) (`boost_1_68_0-msvc-14.1-64` or `-32`). I'm not confident this will change behavior as it's the same version of boost, just built with a different compiler - I'd be surprised if any system calls are changed, but may be worth a try.

**Baud Rate**
The most common cause of communication failures is an incorrectly specified (or unspecified) baud rate. If unspecified MSCL automatically assumes 921600. If connected via USB on Windows the specified baud rate does not matter.

**Bandwidth Issues while Streaming**
This is commonly reported as intermittent or random communication failures but can cause more consistent failures as well. If the device is streaming data and running into bandwidth issues command responses are often dropped, even when just trying to read basic device info or send a ping. Setting the node to idle is the best way to re-establish consistent communication. The below can be added when the user wants to ensure successful communication.
```
// attempt set to idle multiple times in case responses are dropped due to data streaming
for(int i = 0; i < 3; i++)
{
  try
  {
    m_inertial_device->setToIdle();
    break; // stop trying if successful
  }
  catch(const mscl::Error& e){/*ignore*/}
}
```

bharatm11, 2021-06-15:
**SensorConnect** seems to work fine. 

**Boost version**
We are using Boost-1.73.0 with MSCL_52.2.1. On the old environment configuration, we were using Boost-1.72.0. 

**Baud rate**
We are using USB on Windows so, this is not a factor.

**Bandwidth issues while streaming**
` 

    for(int i = 0; i < 3; i++)
    {
      try
      {
        node.setToIdle();
        break; // stop trying if successful
      }
      catch(const mscl::Error& e){
            std::cout << e.what(); /*ignore*/
        }
    }`

The above code prints _"Failed to communicate with the device."_

mglord, 2021-06-25:
We are currently building MSCL with Boost 1.68.0 - clearly it worked for you with 1.72.0 but we have not done any compatibility testing with newer versions. Could you try installing and linking to Boost 1.68.0 (or 1.72.0) to rule out a compatibility issue?

It's encouraging that SensorConnect communicates successfully - it relies on MSCL for device communication so we should be able to get MSCL working.

Thanks for your patience!

closed 2021-06-30 by bharatm11