LORD-MicroStrain/MSCL #165: Register MSCL.dll for LabVIEW compiled executables

aerophd2001, 2021-06-17:
I have a working application using MSCL within the LabVIEW development environment. However, the compiled executable errors on any of the MSCL .NET function calls. I then downloaded the Windows .NET binaries for MSCL, but registering the .dll's using regsvr32 produces the error:

"The module "mscl.dll" was loaded but the entry-point DllRegisterServer was not found. Make sure that "mscl.dll" is a valid DLL OR OCX file and then try again later."

mglord, 2021-06-17:
Can you clarify a few things:
- In the working MSCL application were you using the pre-compiled .NET .dlls or did you build it from source?
- Which function calls specifically fail? Does any reference to MSCL (including instantiating object ie Connection, BaseStation, WirelessNode, etc)? What are the errors?
- "then downloaded the Windows .NET binaries for MSCL" unless you previously built MSCL from source can you point me to what you were using previously and what you're trying to use now?

Regarding `regsvr32`: it looks like this operation requires a .dll to have a single function marked as an entry point which MSCL does not have. You should be able to place the .dlls in a location that LabVIEW will automatically look for them to avoid needing to register them. Is there a specific reason you require registering with `regsvr32`?

I'm not very familiar with LabVIEW - apologies if I've misunderstood!

aerophd2001, 2021-06-17:
> 
> 
> Can you clarify a few things:
> 
>     * In the working MSCL application were you using the pre-compiled .NET .dlls or did you build it from source?
>      A: I have installed SensorConnect and the MSCL LabVIEW libraries from: https://github.com/LORD-MicroStrain/LabVIEW-MSCL (and installed using the JKI VI Package Manager).

>     * Which function calls specifically fail? Does any reference to MSCL (including instantiating object ie Connection, BaseStation, WirelessNode, etc)? What are the errors?
>      A: The very first function that my application needs to call from the MSCL library is the ListBaseStationCOMPorts function which begins with a call to the method Devices/listBaseStations. 

>     * "then downloaded the Windows .NET binaries for MSCL" unless you previously built MSCL from source can you point me to what you were using previously and what you're trying to use now?
>      A: From the Windows / .NET link on: https://github.com/LORD-MicroStrain/MSCL
> 
> Regarding `regsvr32`: it looks like this operation requires a .dll to have a single function marked as an entry point which MSCL does not have. You should be able to place the .dlls in a location that LabVIEW will automatically look for them to avoid needing to register them. Is there a specific reason you require registering with `regsvr32`?
> 
> I'm not very familiar with LabVIEW - apologies if I've misunderstood!

mglord, 2021-06-17:
Okay, that makes sense. Thanks for the info!

Were you able to communicate with a base station/node prior to adding the `Devices::listBaseStations()` function?

I'll need to ask my coworker who put together the LabVIEW-MSCL libraries to get a better understanding of how those are setup to install/include MSCL.

In the meantime, have you tried putting the .dlls in a file location that LabVIEW will [look for them automatically](https://knowledge.ni.com/KnowledgeArticleDetails?id=kA00Z000000PAm6SAG&l=en-US) without registering them with `regsvr32`? I'm unsure, but this may need to be done in a separate project than the one already using the LabVIEW-MSCL libraries.

I'll get clarification from my coworker about our LabVIEW package and hopefully will be able to get back to you with a more comprehensive answer.

aerophd2001, 2021-06-17:
BTW- My hardware for this project: WSDA-200-USB, V-Link-200-OEM, SG-Link-200-OEM.
My application currently encompasses the following functionalities:
- Programmatically find Base Stations
- Programmatically find Nodes on a Base Station
- Configure Nodes (sample rate, voltage range, sample method, etc.)
- Acquire Shunt Calibration on strain gauges
- Stream data (non-sync at the moment but hoping to get sync working at some point)

mglord, 2021-06-18:
We have a C# example (using .NET) for sync streaming [here](https://github.com/LORD-MicroStrain/MSCL/blob/master/MSCL_Examples/Wireless/CSharp/MSCL_Wireless_Example_CSharp/MSCL_Wireless_Example_CSharp/startSyncSampling.cs).

Basically you add all the nodes to a `SyncSamplingNetwork`, MSCL calculates and assigns a transmit schedule for each node based on their individual sampling configuration (channels, rates, etc.), and then starts sampling for all nodes in the network.

Hope this helps, let us know if you run into any issues implementing this!

Any luck with moving the MSCL .dlls to the default LabVIEW search directories?

aerophd2001, 2021-06-18:
Still no luck with the .dll's. Thanks for the Sync example. I'll look into implementing it once I get the .dll problem resolved.

mglord, 2021-06-21:
Would you mind clarifying the following from your original comment:

> I have a working application using MSCL within the LabVIEW development environment. However, the compiled executable errors on any of the MSCL .NET function calls.

Are you experiencing these runtime errors with just the LabVIEW-MSCL package, or did they start only after you downloaded and tried to use the .NET .dlls? Have you at any point been able to communicate with a base station/node via LabVIEW with MSCL?

Is there any associated error message? Did the error change when you moved the .dlls?

Thanks!

aerophd2001, 2021-06-21:
I first attempted simply compiling my application. It failed with LabVIEW Error #403500 at ListBaseStationCOMPorts.vi. As I dug into the code, this was the first of the MSCL functions my application calls. Everything else in the code is dependent on finding the Base Station(s). If no Base Stations are found, none of the rest will do anything.

I then did some research and found these functions came from the "mscl.dll". My assumption was that for some reason, this .dll wasn't fully/properly registered on my machine. That is when I downloaded a fresh copy of the binaries and attempted to use regsvr32, resulting in my original post.

I have since attempted to force the inclusion of the .dll into my application build script by adding it to the "Always Included" section of the "Source Files" page of the LabVIEW Build Specification. This has the result of creating an additional folder called "data" within the build folder, and containing the .dll. This compiled executable errors in the same fashion. I have also tried to simply place the .dll so it resides right beside the compiled application. That too fails.

I have exhausted the permutations and combinations of ideas I could come up with to solve the issue.

mglord, 2021-06-21:
I've heard back from my coworker who helped clarify the process for installing the MSCL LabVIEW package, and the setup and end result of linking it within a LabVIEW project. The below steps are probably review for you, but would you mind confirming them to make sure we're on the same page?

- Download the latest lord_sensing_microstrain_lib_mscl-0.6.1.4.vip package from [LabVIEW-MSCL Releases](https://github.com/LORD-MicroStrain/LabVIEW-MSCL/releases)
- Installing this using the VI Package Manager will result in the `LORD Sensing - MicroStrain\_MSCL_internal_deps\x86` directory added to Program Files folder (example directory for LabVIEW 2012, may differ depending on version: `Program Files (x86)\National Instruments\LabVIEW 2012\vi.lib\LORD Sensing - MicroStrain...`)
- This x86 folder will contain two .dlls: `MSCL.dll` and `MSCL_Managed.dll`
  - To upgrade to the latest version of MSCL, replace these files with the files of the same name from the pre-built MSCL .NET download
- Recommended for convenience: add the installed MSCL x86 directory to the project tree in LabVIEW to make it more easily accessible
- As you mentioned, the x86 folder should be added to the "Always Included" file set under Source Files to ensure LabVIEW moves the .dlls to the data folder and re-links them accordingly when compiled
- Upon building the program, check to confirm both `MSCL` and `MSCL_Managed` dlls have been moved to the data folder - it sounds like you've targeted `MSCL.dll` successfully, but may be missing `MSCL_Managed.dll`

If you've confirmed these steps and it's still throwing errors, can you try hard-coding the base station port and manually creating the `Connection` and `BaseStation` objects and check whether you are able to ping the base station. This will help determine whether the issue is with MSCL or with the `listBaseStations()` function specifically.
```
const string COM_PORT = "COM10";
mscl::Connection connection = mscl::Connection::Serial(COM_PORT, 3000000);
mscl::BaseStation base = mscl::BaseStation(connection);

bool pingSuccess = base.ping(); // true if successful response, false if failure or no response
```

Thanks for your patience while I come up to speed on LabVIEW configuration!

aerophd2001, 2021-06-24:
I have finally gotten my application to compile properly. Here are some of the steps that I performed ultimately leading up to a successful compile:

> test vi:  functioning in labview environment

> compiled test vi:  fails
  error #403500 at listbasestationcomports.vi

> overwrote .dlls in  .../vi.lib/lord sensing - microstrain/dotnet/x86

> added  .../vi.lib/lord sensing - microstrain/dotnet/x86 to labview project

> test vi:  failed in labview environment
  (same error as previous compiled executable)

> uninstalled mscl using jki vi package manager

> reinstalled mscl using jki vi package manager

> test vi:  functioning in labview environment again

> compiled test vi:  functions properly

I'm uncertain exactly which step of the above ultimately lead to the success. I would speculate either the addition of the x86 folder to the project, or more likely, the uninstall/reinstall of the MSCL. Regardless, I'm on to my next task of getting the sync streaming functioning. You can close out this issue now.

mglord, 2021-06-24:
Glad you got it working and thanks for sharing how you resolved the issue!

Let us know if you run into any issues implementing sync streaming!

closed 2021-06-24 by mglord