LORD-MicroStrain/MSCL #166: [Question] Sample burst for 60 seconds only

Testing1User, 2021-06-25:
Hi,

I tried looking at the mscl library but I cant seem to find something with the same function as the burst sampling for certain number of time in sensor connect. Could anyone help me with this enquiry please.

[image]

I am trying to set 4 nodes to sample for 1minute and set to them idle automatically, after it samples for 1 minute. 
There only seems to be an option to start syncSampling with a given timestamp parameter.

Many thanks,

Testing1User, 2021-06-25:
Hi,

I can find the option  to change the node to sync burst mode.
[image]

Do I have to manually set the number of sweeps I want to sample for with unlimited duration disabled? 

Thanks,
Ash

mglord, 2021-06-25:
Sorry for the confusion! You're correct - in MSCL the way to set limited duration sampling is specifying the number of sweeps to collect. SensorConnect calculates this value behind-the-scenes based on the sample rate and specified time. Note: the limited sweep count must be a multiple of 100 - this is why the specified time in SensorConnect is often adjusted once entered.

Just to clarify `samplingMode_syncBurst` only needs to be used if you want the limited sampling duration to repeat periodically ('bursting every' checkbox checked in SensorConnect UI) - if you just want the sampling duration to run once the mode should be set to `samplingMode_sync`.

Hope this helps, let us know if additional clarification is needed!

Configuration summary:
```
mscl::WirelessNodeConfig config;
        
// set the duration in sweep count (must be a multiple of 100)
// sweeps in time duration = sample rate (sweeps/s) * duration (s)
uint32_t sweeps = 256 * 30;

// adjust to be an even 100
sweeps = sweeps / 100;
sweeps += 1; // remove this if you want to adjust down instead of up
sweeps *= 100;
config.numSweeps(sweeps);

// important:
config.unlimitedDuration(false);

// Set to samplingMode_sync to just sample for the specified duration once
config.samplingMode(mscl::WirelessTypes::samplingMode_syncBurst);
config.sampleRate(mscl::WirelessTypes::sampleRate_256Hz);

// only needed if sampling mode is syncBurst
config.timeBetweenBursts(mscl::TimeSpan::Seconds(30));

node.applyConfig(config);
```

Testing1User, 2021-06-29:
Hi mglord,

That works great.
Thank you for your help!

closed 2021-06-29 by Testing1User

Testing1User, 2021-06-29:
Hi @mglord ,

Another quick questions when you are available - 

I use LXR mode mainly, if in the case my nodes are in LXRS+ I want to set them to LXRS mode again.

In order to set my nodes to LXRS mode do I need to create a baseStation config object and set its configutation to LXRS+ to then communicate with my node, set it to LXRS and then set the baseStation config back to LXRS mode?
This is only the current way I can do it. Just trying to replicate SensorConnects quick click of a button to change between the two different communication protocols for a given node.

Thanks

mglord, 2021-06-29:
I'm glad you were able to get the limited duration sampling working!

Regarding switching between communication protocols (LXRS, LXRS+) your current strategy is correct - the base station can only talk to the node with the same configured protocol, even to switch the node's configured protocol. In SensorConnect we do exactly what it sounds like you've implemented: switch the base station protocol (LXRS -> LXRS+), change the node's (LXRS+ -> LXRS), and then switch the base station back (LXRS+ -> LXRS).

**Optional:** There is a function to test a communication protocol on a node prior to actually setting it, `WirelessNode::testCommunicationProtocol` - SensorConnect uses this every time the node protocol is changed, but it is most important when switching from LXRS to LXRS+ due to the reduced range of LXRS+. Basically it temporarily switches the base and node protocol then attempts to ping the node to ensure that communication will not be lost when the protocol is changed. The function returns false if the ping fails, true on success. Regardless of ping success both the base station and node will automatically switch back to the original protocol. The base station and node need to start on the same protocol to use this function.

To summarize, the SensorConnect workflow when the node comm protocol is changed:
- Initial config:
  - base station: LXRS
  - node: LXRS+
- Set base station to LXRS+
- Test node for LXRS (`node.testCommunicationProtocol(mscl::WirelessTypes::CommProtocol::commProtocol_lxrs`)
- If successful, set node to LXRS
- Set base station back to LXRS

Definitely let us know if you're running into issues with this workflow, but it sounds like your implementation is correct!

Testing1User, 2021-07-09:
Hi,

Thank you for the very useful tips.
They work great and I have everything running. The testCommunicationProtocol is very clever and works great!

Cheers

mglord, 2021-07-09:
Excellent! Glad you were able to get everything working!