LORD-MicroStrain/MSCL #168: Debian buildscript not working anymore

AckriteTJ, 2021-07-01:
Can you please check the linuxManualScript for Debian? It’s not working for me anymore since last update of mscl

mglord, 2021-08-10:
Can you provide any additional details about how it is failing? Any error messages?

shirmg, 2021-09-08:
I believe I am running into the same issue with the Debian build scripts.
I built for Python.  
After going through the build process, I got the following error when my script tried to import MSCL:

[image]

Please let me know what other information I can provide.  I have successfully used the pre-built python version of MSCL for Debian, but I need to build for an ARM v6 processor.  Thanks

mglord, 2021-09-10:
Thanks for the info and sorry for the inconvenience! We'll look into this and let you know what we find/when it's resolved!

closed 2024-10-24 by github-actions[bot]