LORD-MicroStrain/MSCL #171: [Question] Timestamp Time-Zone Query

Testing1User, 2021-07-09:
Hi,

Would like some help on the nodes Timestamp please!

**What I am doing** 
I am currently logging data to my G-Link-200 Nodes and downloading the data at a later time. Saving the data & timestamp into a .csv
**How I am doing it**
I get the timestamp by accessing data.timeStamp() function where 'data' represents a data sweep from getNextData().
**My set up**
Timezone - British summer time (BST) using Raspberry Pi Debian mscl - Python.
**Query**
Timestamp is set as UTC in the nodes - is it possible to make it sync with the system time zone (BST)? 
e.g at 15:30(BST) system time, my downloaded data shows - node was running at 14:30(UTC).

[image]
**Example - Picture 1** 
1. Represents time from the system
2. Actual time saved in the G-Link/Basestation? Not sure how this works.

I am trying to set time 2 to be the same as time 1.
I have tried using timestamp.setTimeNow() - but that has no affect on it - could it be the way my system is set up? 

Any guidance or tips would be appreciated.
Cheers!

mglord, 2021-08-10:
Unfortunately there is no way to set the nodes' internal time - they will always transmit timestamps in UTC.

Using `setTimeNow()` does not impact the timestamp on the node, if run on the actual `DataSweep`'s timestamp it would just overwrite the actual sampled time with the current system time - I can't imagine a scenario where this would the desired result.

The easiest option is probably to convert the UTC timestamp to your timezone prior to writing it to csv.
Example: 
```
// get timestamp from DataSweep instead of using setTimeNow()
mscl::Timestamp sweepTs;
sweepTs.setTimeNow();

cout << endl << "Timestamp UTC: " << sweepTs.str() << endl;

// BST (+1)
// 3600000000000 nanoseconds/h
mscl::Timestamp adjusted(sweepTs.nanoseconds() + (1 * 3600000000000));

cout << endl << "Timestamp BST: " << adjusted.str() << endl;
```

Hope this helps!

mglord, 2021-08-23:
I'm going to go ahead and close this, but if you run into any problems feel free to re-open this or another issue! Thanks!

closed 2021-08-23 by mglord