LORD-MicroStrain/MSCL #172: macOS / OSx support

kopecn, 2021-07-17:
Andrew Deering asked me to relay this question to this forum

Andrew,
 
Thanks for responding.  I did review the github page and needed a bit of help in one place. 
 
On the profile page for MSCL it states that this can be run on OSx and was looking around on git for the OSx instructions.  If possible can you provide me a pointer to how to get MSCL operational on OSx.  BTW, the rest of the python stuff and the API documentation is most excellent.  
 
Much Appreciated. 
Nick

mglord, 2021-08-10:
I have not personally built/run MSCL on OSx, but I'm pretty sure it has been done successfully. I'd like to look further into this to put together an instruction page, but in the meantime the [linux manual build instructions](https://github.com/LORD-MicroStrain/MSCL/blob/master/BuildScripts/buildReadme_Linux.md) would be the place to start.

In the build script ([MSCL_Install.py](https://github.com/LORD-MicroStrain/MSCL/blob/master/BuildScripts/linuxManualBuild/Debian/MSCL_Install.py)) you will run into issues with the apt, apt-get calls to install dependencies in the update function, but if you can install those independently and then remove the call to the update function it may just work from there.

If you try it I'd be interested to hear how it goes, and will of course be happy to help work through any issues that may come up!

Sorry for the inconvenience!

mglord, 2021-08-23:
I'm going to go ahead and close this, but feel free to re-open this or open a new issue if you run into any problems! Thanks!

closed 2021-08-23 by mglord