LORD-MicroStrain/MSCL #173: How to find  peak value of accelerometer from zero reference

nayanagg, 2021-07-19:
Hi,
I'm using a 3DM-CV5-10 IMU sensor. I need to find the peak value (reference from zero) from the accelerometer when the acceleration occurs.

mglord, 2021-08-10:
To clarify - are you requesting a function to find the peak value on a channel within a previously collected data set? Unfortunately MSCL doesn't have any built-in support for data post-processing - it is up to users to implement data analysis functions themselves.

Let me know if I've misunderstood your question, otherwise sorry for the inconvenience!

nayanagg, 2021-08-11:
No,
Actually i want to measure the peak values from accelerometer reading in real time like in sinusoidal wave (1 cycle completes  when 1 crest and 1 trough occurs) so, i want to know the peak value of  crest and trough in 1cycle.

closed 2021-08-11 by mglord