LORD-MicroStrain/MSCL #174: How to improve the download rate of the DataLogDownloader on a WirelessNode

SensicalFVO, 2021-08-05:
Hi,

the rate downloading the logged data on a wireless node using SensorConnect is significantly faster compared to the rate using the following code in Matlab:
```
downloader = mscl.DatalogDownloader(node{1});

% Progress bar 
wb = waitbar(downloader.percentComplete()/100,['Downloading logged data from node: ', char(node{1}.name)]);

% sweep counter
s = 1;

% preallocation
downloadedSweeps = cell(2^21,1);

% downloading process in a while loop                
while ~downloader.complete()
     % getNextData() throws an error from time to time -> try statement. 
     try
          downloadedSweeps{s} = downloader.getNextData();
          % update counter and progress bar, when getNextData() has been sucessfull
          s = s + 1;
          waitbar(downloader.percentComplete()/100,wb);
     catch me
          % in case getNextData() fails, retry
          continue
     end
end

% shrink preallocated cell array to the actual number of sweeps
num_sweeps = s - 1;
downloadedSweeps = downloadedSweeps(1:num_sweeps);
```
Although no other operations are performed, which could slow down the downloading process, the download is approximately five times slower than a download of the same logged data using SensorConnect. Overall the code is working, I am just trying to improve the rate. Is there a way to do that?

Thank you!

mglord, 2021-08-10:
Can you try removing the `waitbar()` update function? I'm assuming this ends up drawing to the screen which tends to be slow.

SensicalFVO, 2021-08-13:
Hello mglord,

thank you for your answer! Removing the waitbar(), has solved my problem. The above code without the `waitbar` is only 20% slower compared to SensorConnect, now. I was wondering if there is a way to increase the signal strength of the wireless node, because the `getNextData()` seemed to fail quite often. May be this will overcome the remaining 20% in performance lack.

mglord, 2021-08-23:
I'm glad that worked and you're seeing a significant improvement!

The transmit power on both the base station and node are configurable, see example code below. SensorConnect does not change the configuration prior to downloading logged data, but it seems that increasing the transmit power could lower the failure rate, reducing the need for retries. It's possible that your runtime environment may just have a bit more overhead than SensorConnect.

Hope this helps!

```
// config either BaseStationConfig or WirelessNodeConfig object
config.transmitPower(mscl::WirelessTypes::TransmitPower::power_16dBm);

// device either BaseStation or WirelessNode object
device.applyConfig(config);
```