LORD-MicroStrain/MSCL #176: Apply low pass filter at 10 Hz for all axis reading

nayanagg, 2021-08-12:
Hi,

Kindly let me know how to apply 10Hz low pass filter for all the axis readings via MSCL in python.

mglord, 2021-08-12:
You can configure the device to automatically apply a configurable low-pass filter to the scaled accel, gyro, mag, and pressure data with the `InertialNode::setAdvancedLowPassFilterSettings` function. This function accepts a vector of `AdvancedLowPassFilterData` objects that contain the desired settings for each filter you want applied. If you are using the CV5-10 then only the accel and gyro channels/filters are available.

The below example code configures the device to apply both accel and gyro low-pass filters at a 10 Hz cutoff frequency. It's in C++ but all object, property, and function names are the same in Python. Let me know if you run into any trouble!

```
// set up desired accel low-pass filter config
mscl::AdvancedLowPassFilterData accelFilter;
accelFilter.dataDescriptor = mscl::MipTypes::ChannelField::CH_FIELD_SENSOR_SCALED_ACCEL_VEC;
accelFilter.applyLowPassFilter = true;
accelFilter.manualFilterBandwidthConfig = mscl::AdvancedLowPassFilterData::USER_SPECIFIED_CUTOFF_FREQ;
accelFilter.cutoffFrequency = 10;

// set up desired gyro low-pass filter config
mscl::AdvancedLowPassFilterData gyroFilter;
gyroFilter.dataDescriptor = mscl::MipTypes::ChannelField::CH_FIELD_SENSOR_SCALED_GYRO_VEC;
gyroFilter.applyLowPassFilter = true;
gyroFilter.manualFilterBandwidthConfig = mscl::AdvancedLowPassFilterData::USER_SPECIFIED_CUTOFF_FREQ;
gyroFilter.cutoffFrequency = 10;

// apply low-pass filter config(s) to device
// Note: device is an InertialNode instance
device.setAdvancedLowPassFilterSettings({ accelFilter, gyroFilter });
```

nayanagg, 2021-08-12:
Thanks for the support.

also on applying a 10 Hz low pass filter to accelerometer readings,  will it starts to attenuate the reading from 7 Hz?

closed 2024-10-24 by github-actions[bot]