LORD-MicroStrain/MSCL #177: Gateway real time data export

guillegago, 2021-08-18:
Hello. Hope someone can help me with this issue on a project or direct me to the correct channel.
Using G-LINK 200 8G and WSDA, does the WSDA send via ethernet or other (in real-time) the acquired data from accelerometers to other systems as a PID inputs for example to compensate vibrations? If yes, how we can do that by using your software tools?

Could you please give me some details of the libraries/API to be used to accomplish with this requirement?
If there is some other way to accomplish with the requirement please let me know.

Appreciated for the feedback.

guillegago, 2021-08-18:
Following my request above, does anyone know which are the LORD drivers for Labview?

mglord, 2021-08-23:
MSCL is used to collect and parse data from raw byte strings to a more easily readable format - what you do with the data from there is up to you. You didn't specify which model WSDA you're using, but if you have the WSDA-2000 this is designed to automatically send data to our cloud service, [SensorCloud](https://sensorcloud.com/), which you could pull data from via our [SensorCloud SDK](https://github.com/LORD-MicroStrain/SensorCloud).

You can find our MSCL-based LabVIEW package [here](https://github.com/LORD-MicroStrain/LabVIEW-MSCL).

If you need help reading your data from a device or SensorCloud we can definitely help with that, but we do not offer tools to reformat or send the data to different systems.

guillegago, 2021-08-24:
Crystal clear, thank you. Gateway is supposed to be the WSDA-2000. Our purpose is definitely to use MSCL-Labview to collect and process data from wireless Lord Microstrain equipments.