LORD-MicroStrain/MSCL #178: data collecting with an unknown node and data type errors

nesslili, 2021-08-20:
Hello,

I am using wireless data collection. my node address is 58700. however when data collecting, there is always another node with address 4660 there with errors. but no information for 58700. please see below output:

Packet Received:  Node 4660 Timestamp: 2021-08-20 21:13:31.873037400 Tick: 0 Sample Rate: 1Hz Base RSSI: 999 Node RSSI: 999 DATA: beaconEcho  datatype: 10Error: Data was accessed using the wrong data type.
Press any key to continue . . .

mglord, 2021-08-23:
Are you able to ping node 58700? If not, please confirm that the node and base station are communicating on the same frequency - the easiest way to do this is via SensorConnect, but if you need to do this via MSCL let us know and we can point you in the right direction.

Regarding node 4660 - if you want to continue streaming data from this node within range of the base station but do not want this base station to receive it you will need to change the base station to a different frequency. The error looks like it is not an actual error within the data, but just using an incompatible function to access it. Where you are printing the value, can you try using the function `point.as_string()` where `point` is a `WirelessDataPoint` object - if I'm not mistaken the `as_string()` function should handle printing any of our available data output types.

I hope this helps - let us know if further clarification is needed!

mglord, 2022-01-04:
I'm going to go ahead and close this, but if you need further assistance please feel free to reopen it! Thanks!

closed 2022-01-04 by mglord