LORD-MicroStrain/MSCL #185: We can not get data stream from G-Link-200

Panying005, 2021-09-28:
We using the C# for coding as the MSCL sample code instruction. We just copy the sample code.
We had set up the connection for the network system and we are able to Ping to Node, Set Node to Idle. But we can not get data from the X,Y,Z axis of G-Link-200.
Is there anything wrong as below of my code?  

//create a SyncSamplingNetwork object, giving it the BaseStation that will be the master BaseStation for the network
mscl.SyncSamplingNetwork network = new mscl.SyncSamplingNetwork(baseStation);

//add a WirelessNode to the network.
//Note: The Node must already be configured for Sync Sampling before adding to the network, or else Error_InvalidNodeConfig will be thrown.
network.addNode(node);

network.ok();               //check if the network status is ok
network.lossless(true);     //enable Lossless for the network
network.percentBandwidth(); //get the total percent of bandwidth of the network

//apply the network configuration to every node in the network
network.applyConfiguration();

//start all the nodes in the network sampling.
network.startSampling();

//create the BaseStation, passing in the connection
mscl.BaseStation basestation = new mscl.BaseStation(connection);

while(true)
{
    //get all the data sweeps that have been collected, with a timeout of 500 milliseconds
    mscl.DataSweeps sweeps = basestation.getData(500);

    foreach(mscl.DataSweep sweep in sweeps)
    {
        sweep.nodeAddress();    //the node address the sweep is from
        sweep.timestamp();      //the TimeStamp of the sweep
        sweep.tick();           //the tick of the sweep (0 - 65535 counter)
        sweep.sampleRate();     //the sample rate of the sweep
        sweep.samplingType();   //the SamplingType of the sweep (sync, nonsync, burst, etc.)
        sweep.nodeRssi();       //the signal strength at the Node
        sweep.baseRssi();       //the signal strength at the BaseStation
        sweep.frequency();      //the radio frequency this was collected on

        //get the vector of data in the sweep
        mscl.ChannelData data = sweep.data();

        //iterate over each point in the sweep (one point per channel)
        foreach(mscl.WirelessDataPoint dataPoint in data)
        {
            dataPoint.channelName();    //the name of the channel for this point
            dataPoint.storedAs();       //the ValueType that the data is stored as
            dataPoint.as_float();       //get the value as a float
        }
    }
}

mglord, 2021-10-13:
@vinod743374 - answered on #133. In the future please open a new issue when you have a different question than the existing topic and are not the original poster, it helps keep each issue single-threaded. Thanks!

mglord, 2022-01-04:
I hope you were able to get your G-Link data successfully streaming! I'm going to close this, but if the issue persists or you have any additional questions feel free to reopen this or another issue! Thanks!

closed 2022-01-04 by mglord