LORD-MicroStrain/MSCL #186: Is there any way to get the X Y Z channel data separately？

wuliyangyang, 2021-09-29:
I used your demo code, but I didn't find the corresponding method. What I think is that there should be a method like this: "double[] ch1Datas = node.GetCh1DataStream()"to get  X channel data.

mglord, 2021-09-30:
All data read from the base station is returned from the `baseStation.getData()` function as a collection of `DataSweep` objects which can be identified by node address and contain collections of `WirelessDataPoint` objects (`dataSweep.data()`) that can be identified by channel id.

If you would like to create your own functionality that retrieves and stores the data from the `BaseStation` object and provides access to filtered sub-sets of data like you're describing I would recommend looking at the [parseData](https://github.com/LORD-MicroStrain/MSCL/blob/master/MSCL_Examples/Wireless/CSharp/MSCL_Wireless_Example_CSharp/MSCL_Wireless_Example_CSharp/parseData.cs) file within the example project and `DataSweep` object [documentation](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/DataSweep.h:DataSweep) to get a better understanding of the returned data structure.

Hope this helps!

wuliyangyang, 2021-10-03:
> All data read from the base station is returned from the `baseStation.getData()` function as a collection of `DataSweep` objects which can be identified by node address and contain collections of [...]

Could you Please exactly give a code example to get the data of X Y X channel? I don’t know much about your code structure

mglord, 2022-01-04:
I hope you where able to get the data into the format you were hoping! I'm going to go ahead and close this, but if you continue to run into trouble or have any additional questions feel free to reopen this or another issue! Thanks!

closed 2022-01-04 by mglord

dheazkelvin, 2022-11-22:
what about 2 nodes? how to get separates data from multiple node?