LORD-MicroStrain/MSCL #196: mscl.Connection.Serial over RPi USB Port

dmiles, 2021-11-05:
Is there a tech note on using the mscl python lib with a USB port on a raspberry Pi4?  I have a 3DM-GQ7 plugging into the USB port and it powers up fine. I can see the the port by running the lsusb command in terminal.
        Bus 001 Device 094: ID 0483:5740 STMicroelectronics Virtual COM Port

I tried to use a symlinks rule to map the usb port but without success. Cannot get connected
      SUBSYSTEM=="tty", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="5740", 
      SYMLINK+="ttyUSBGQ7", GROUP="dialout", MODE="0666"

connection = mscl.Connection.Serial("ttyUSBGQ7", 115200)
gives me a runtime exception of invalid Com port.

Has anyone done this? Any help would be appreciated.

mglord, 2021-11-11:
Currently to use symlinks with MSCL you need to use the realpath function to resolve the actual port name at runtime. If this doesn't resolve the issue, can you remove the symlink and pass in the actual port name to confirm that you are able to connect to the GQ7 otherwise? That will help narrow down whether the issue is actually in creating the connection or it's an issue with the symlink.

Adding the following worked for me, let us know if this workaround doesn't resolve the issue!

Add to includes:
`#include <stdlib.h>`

Determine actual port:
```
const string COM_PORT = "/dev/bestGq7";
const int BAUD = 115200;

char* actualPort = realpath(COM_PORT.c_str(), NULL);
string portString(actualPort);
cout << "Attempt create connection: " << COM_PORT <<" (" << portString << "), " << BAUD << endl;

//create a SerialConnection with the COM port
mscl::Connection connection = mscl::Connection::Serial(portString, BAUD);
cout << "Connection created" << endl;
```
Output:
```
Attempt create connection: /dev/bestGq7 (/dev/ttyUSB1), 115200
Connection created
```

mglord, 2022-01-04:
I hope you were able to get this working! I'm going to go ahead and close this but feel free to reopen if you continue to experience this issue! Thanks!

closed 2022-01-04 by mglord