LORD-MicroStrain/MSCL #197: void lostBeaconTimeout( uint16 minutes )

RonNeal, 2021-11-11:
Please let me know what happens when "lostBeaconTimeout" is executed. On my G-Link-200-40G-OEM I have the value set to two in the config code. I assumed that two minutes after the beacon has been lost the node would either be set to idle or put to sleep which is not the case.

mglord, 2021-11-11:
When the lostBeaconTimeout is set the node simply stops sampling and transmitting after the configured amount of time has passed without a beacon. It stays in sampling mode but saves power because it is no longer actually sampling. This allows the unit to resume sampling once the beacon has returned. There may be some buffer period between the beacon returning and sampling resuming, but I would have to get clarification from someone more familiar with that specific functionality regarding why and how long you might expect to wait for data transmission to resume.

Sorry for the confusion, let us know if further clarification is needed!

mglord, 2022-01-04:
I'm going to close this, but if you have any additional questions please feel free to reopen it! Thanks!

closed 2022-01-04 by mglord