LORD-MicroStrain/MSCL #198: Segmentation fault" error in RaspberryPi 3B+

Creact-Onizuka, 2021-11-16:
Hi
I'm trying to use RaspberryPi 3 model B+ and MSCL, but it showed and does not work with "segmentation fault" error.
and I have also tried pre-built package and source code compilation, but both are not going well.
I'm using version 62.1.2. Would you please advise me how to solve this matter
[image]
.

robbiefish, 2021-11-23:
Hi @Creact-Onizuka

A couple questions:
1. What architecture is the OS you are using? This can be found with `uname -m`
2. Which package are you installing? the `armhf` or `arm64` package?
3. Are you seeing a segmentation fault with both the package and source install?
4. Did you make any modifications to the example source code?

I unfortunately do not have access to a RaspberryPi3 to test this with, but I tested a similar setup in `arm32v7` and `arm64v8` docker containers, and was unfortunately unable to reproduce the error you were seeing.

Could you try running the program in `gdb` and sending the core dump? Instructions for reference:
```
$ gdb ./YourProgram
(gdb) r
(gdb) gcore
```

Creact-Onizuka, 2021-11-26:
Thanks for the response @robbiefish

1.
armv7l
I used the following software to perform a standard setup.
https://www.raspberrypi.com/software/

2.
package is armhf
https://github.com/LORD-MicroStrain/MSCL/releases/download/v62.1.2/c++-mscl_62.1.2_armhf.deb

3.
Yes. The source code was downloaded from here.
https://github.com/LORD-MicroStrain/MSCL/archive/refs/tags/v62.1.2.zip
The build runs the following script.
MSCL-62.1.2\BuildScripts\linuxManualBuild\Debian\MSCL_Install.py
Compilation stopped halfway through, so I changed the swap size and cooled the CPU.

sudo service dphys-swapfile stop
sudo vim /etc/dphys-swapfile
	CONF_SWAPSIZE=1024
sudo service dphys-swapfile start

The build completed successfully, but when I ran it, the result was the same as the package.

4.
Change the 15th line of the sample code as shown below.
    const string COM_PORT = "/dev/ttyACM0";

As seen in the previous image, I added -pthread to the compile options and set the path to libmscl.so before execution.

The results of the dump and main.cpp are attached.

Thanks.
[mscl-raspi_dump.zip](https://github.com/LORD-MicroStrain/MSCL/files/7606817/mscl-raspi_dump.zip)

Creact-Onizuka, 2021-11-26:
Core dump message.

Reading symbols from ./YourProgram...(no debugging symbols found)...done.
(gdb) r
Starting program: /home/pi/mscl/MSCL_Inertial_Example_C++/YourProgram 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0x767b7450 (LWP 1084)]
[Thread 0x767b7450 (LWP 1084) exited]
Node Information: 

Thread 1 "YourProgram" received signal SIGSEGV, Segmentation fault.
0x76d9c4b8 in mscl::Connection::write(mscl::ByteStream const&) const ()
   from /usr/share/c++-mscl/libmscl.so
(gdb) gcore
Saved corefile core.1080

Rejas181, 2024-04-17:
Did you solved it?, Because I'm having the exact same Issue

Creact-Onizuka, 2024-04-18:
This problem could not be solved and switching to a 64-bit environment worked.

closed 2024-04-18 by Creact-Onizuka