LORD-MicroStrain/MSCL #199: mscl for burst mode

nesslili, 2021-11-17:
Can you provide an example how to code burst mode in mscl to realize: Every 60 seconds you will get a 0.390625 seconds of 512 Hz data. 

Thanks,
Lili

mglord, 2022-01-05:
Sorry for the delay!

In MSCL the way to set limited duration sampling is specifying the number of sweeps to collect. SensorConnect calculates this value behind-the-scenes based on the sample rate and specified time. Note: the limited sweep count must be a multiple of 100 - this is why the specified time in SensorConnect is often adjusted once entered. The code snippet below has the calculation based on time and sample rate, but you can also just specify the `sweeps` value.

Just to clarify samplingMode_syncBurst only needs to be used if you want the limited sampling duration to repeat periodically ('bursting every' checkbox checked in SensorConnect UI) - if you just want the sampling duration to run once the mode should be set to samplingMode_sync.

Hope this helps, let us know if additional clarification is needed!

Configuration summary:
```
mscl::WirelessNodeConfig config;

//set the sample rate
config.sampleRate(mscl::WirelessTypes::WirelessSampleRate::sampleRate_512Hz)
        
// set the duration in sweep count (must be a multiple of 100)
// sweeps in time duration = sample rate (sweeps/s) * duration (s)
uint32_t sweeps = 512 * 0.390625;

// adjust to be an even 100
sweeps = sweeps / 100;
sweeps += 1; // remove this if you want to adjust down instead of up
sweeps *= 100;
config.numSweeps(sweeps);

// important:
config.unlimitedDuration(false);

// Set to samplingMode_sync to just sample for the specified duration once
config.samplingMode(mscl::WirelessTypes::samplingMode_syncBurst);
config.sampleRate(mscl::WirelessTypes::sampleRate_256Hz);

// only needed if sampling mode is syncBurst
config.timeBetweenBursts(mscl::TimeSpan::Seconds(30));

node.applyConfig(config);
```

mglord, 2022-01-21:
I hope you were able to get this working! I'm going to close this issue, but feel free to re-open it if you have any further questions. Thanks!

closed 2022-01-21 by mglord