LORD-MicroStrain/MSCL #200: how to get line position and line speed by using 3dm-gx5-25?

ch1102256774, 2021-12-02:
Hello, I want to know whether the data packets of the sensor **3dm-gx5-25** has **line position and line speed data**? If so, how to get them? If not, which sensor can get these data? Many thanks for help

mglord, 2022-01-05:
Sorry for the delayed response!

I could be mistaken, but it's my understanding that the GX5-25 does not output any position or velocity data, but our GNSS/INS systems do - you can find a feature comparison of our inertial products here: [MicroStrain Inertial Sensors](https://www.microstrain.com/inertial-sensors/all-sensors)

I would recommend reaching out to our support team to get in touch with someone who has more specific product feature knowledge - they'll be able to provide better assistance in determining if the GX5-25 has the functionality you need or, if not, finding the right product to fit your application. You can find our support team contact form here: [Contact Support](https://www.microstrain.com/contact-support)

Hope this helps!

mglord, 2022-01-21:
I'm going to close this issue, but feel free to re-open it if you have any additional questions! Thanks!

closed 2022-01-21 by mglord