LORD-MicroStrain/MSCL #202: Modifying SensorConnect using python or C++ to create a custom dashboard

bcicerone, 2021-12-17:
I Currently have the following hardware from Lord: Node V-Link-200, WSDA-200-USB Base Station, TC-link-200, and G-Link-200-8G

I will be using strain gauges to monitor loads of a helicopter blade for flight testing. The flight engineer must be able monitor loads in real time during flight and if a load is exceeded I want an indicator to pop red so it is easily visible to the flight engineer that a load has been exceeded. I can not do this in SensorConnect with the default settings. 

I am trying to figure out if it is possible to modify SensorConnect to create a custom dashboard?
 
If I use Python or C++ is it possible to manipulate/customize a dashboard in SensorConnect where I would still use the data repo’s in SensorConnect? This way I could stay using SensorConnect Instead of setting up a third party daq and data viewing.

Another route I was thinking about taking is LabVIEW. From my understanding using LabVIEW, I would connect the wireless nodes to LabVIEW for daq and monitoring live data readings. Therefore not using SensorConnect. I'm not sure if using LabVIEW would be the best way to go for my situation. If you can help point me in the right direction. 

Thank you

mglord, 2021-12-17:
In SensorConnect one of the dashboard widgets is a Status Indicator - this widget displays a circle which changes colors based on user-defined value ranges. Once the widget is added to the dashboard when you select it you will see the Channels list on the left side of the screen - in that same panel there is a Settings tab that allows you to assign colors to value ranges:

[image]

It sounds like this could be configured to meet your needs, but definitely let me know if I've misunderstood the functionality you're looking for! There's not really any way to modify SensorConnect - you can certainly build your own interface that uses MSCL to parse data (we have an [MSCL LabVIEW package](https://github.com/LORD-MicroStrain/LabVIEW-MSCL) as well) but I hope we can just find a solution for you within existing SensorConnect functionality!

bcicerone, 2021-12-17:
I have tried using the indicator but it only indicates (changes color) when out of the range, then it will return back to the color selected in range. I need an indication to stay red when the range is exceeded, making it easy for monitoring. The flight test engineer will need to monitor many gauges at a time and it is possible to miss an oscillating load that exceeds the allowable limits. 

Is there a way to setup an indication to pop red and stay red when a load is exceeded in SesnorConnect?

bcicerone, 2021-12-17:
Or if it is possible to make the cell in the numeric display widget pop red when the load is exceeded? I would like to use the numeric display widget rather than the indicator widget.

bcicerone, 2021-12-21:
I have tried using the indicator but it only indicates (changes color) when out of the range, then it will return back to the color selected in range. I need an indication to stay red when the range is exceeded, making it easy for monitoring. The flight test engineer will need to monitor many gauges at a time and it is possible to miss an oscillating load that exceeds the allowable limits.

Is there a way to setup an indication to pop red and stay red when a load is exceeded in SesnorConnect?

 

Or if it is possible to make the cell in the numeric display widget pop red when the load is exceeded? I would like to use the numeric display widget rather than the indicator widget.

 

 <https://user-images.githubusercontent.com/47668674/146596129-11c8827c-9c29-42e4-8b1c-902227994f7e.png>

mglord, 2022-01-05:
Unfortunately we don't currently have a solution for this in SensorConnect. Just out of curiosity, for how long would you want the indicator to stay red - if the threshold had been crossed within the last X amount of time? Until it was manually cleared? Until SensorConnect is relaunched?

bcicerone, 2022-01-05:
The indicator could be turned off when sampling is complete or about 45 min if you wanted to use a time variable. I would want to turn off the indication manually when the exceeded load is reviewed. I need to monitor loads for helicopter blade inflight I want the flight engineer to easily see that the load was exceeded and not have the indication disappear.

If there is no solution in SensorConnect, could I choose to connect LabVIEW to the Lord system using a WSDA-200-USB gateway and use another gateway to connect to sensor connect to be used as a DAQ. Where LabVIEW will be my inflight dashboard and sensorConnect to collect data, is this possible?

mglord, 2022-01-05:
That makes sense, thanks for the feedback!

As far as I'm aware a two-WSDA setup like that would work fine - as long as both are on the same frequency and within range they will both pick up the data transmitted by the node(s). One important consideration with this setup is that only one WSDA should have the beacon enabled at any given time. The beacon is used for timestamping and transmission scheduling in a synchronized network - if multiple are enabled on the same frequency things can get very weird. I don't know that a beacon is required at all in a single-node non-synchronized sampling scenario.

For more network setup information I would encourage you to reach out to our support team - they have more specific product knowledge and will be able to better assist you with questions about our wireless products: [Contact Support](https://www.microstrain.com/contact-support)

Hope this helps!

bcicerone, 2022-01-05:
Is it fairly simple to connect LabVIEW to the Lord gateway? and do you know if there is a better way for what I am trying to achieve? 

Or if there is a way to have live reading in Excel (using Excel as a dashboard) instead of using LabVIEW and still collect data using SensorConnect?

I'm trying to get on the correct path for before purchasing more software and I want to make sure everything will work. 
Your feed back has been hewlpful, Thank You!

mglord, 2022-01-06:
We have many customers who use LabVIEW with MSCL successfully - we offer a LabVIEW VI package ([LabVIEW-MSCL](https://github.com/LORD-MicroStrain/LabVIEW-MSCL)) but I think you can also import the [.NET library](https://github.com/LORD-MicroStrain/MSCL#pre-built-mscl-binariespackages-v6212) manually if you prefer.

I don't have any experience or knowledge of using Excel in that way - if it  is possible I don't see why you wouldn't be able to source the information from MSCL somehow. I'd be very interested to hear if you could get that working, but we would not be able to offer much in the way of support for that integration.

MSCL should work with any tool/project that can consume C++, .NET, or Python libraries. I can't recommend any specific interface utility, but depending on how robust you want the UI to be there may be easier/free solutions out there - if you really just want to display a red icon when a threshold has been crossed it seems like LabVIEW might be overkill. That said, the best solution is often the one that is readily available and you're most comfortable developing in!

mglord, 2022-01-21:
I'm going to go ahead and close this, but feel free to re-open it if you have any additional questions. Good luck with your implementation! Thanks!

closed 2022-01-21 by mglord