LORD-MicroStrain/MSCL #203: WSDA-2000 Internal Temperature

christophjurczyk, 2021-12-17:
Hi,

Is there any way to get the internal temperature of the WSDA-2000 gateway via MSCL?

Thanks in advance!

mglord, 2022-01-06:
Yes, the WSDA-2000 reports its internal temperature as part of the diagnostics packet. To identify the internal temperature value when reading in data with `BaseStation::getData()` check the following:
- the `DataSweep::samplingType()` is `DataSweep::samplingType_Diagnostic`
- the `WirelessDataPoint::channelId()` is `WirelessChannel::ChannelId::channel_diag_internalTemp`

Hope this helps!

christophjurczyk, 2022-01-14:
Hi @mglord,
Thanks for the answer. I am not yet successful. Just to be sure...I would like to read the internal temperature of the WSDA-2000 which I can see in the WSDA Control Panel. Not the internal temperature of a node.

My code:
```
import sys
sys.path.append('/usr/share/python3-mscl/')
import mscl

base_address = "192.168.0.20"
base_port = 5000

connection = mscl.Connection.TcpIp(base_address, base_port)

baseStation = mscl.BaseStation(connection)

print("Connected to base station!")

while True:
    sweeps = baseStation.getData(500)

    for sweep in sweeps:
        if sweep.samplingType() == mscl.DataSweep.samplingType_Diagnostic:
            print("Packet Received")
            print("Node", sweep.nodeAddress())
            print("Timestamp", sweep.timestamp())
            print("DATA:")

            for dataPoint in sweep.data():
                if dataPoint.channelId() == mscl.WirelessChannel.channel_diag_internalTemp:
                    print(dataPoint.channelName(), ":", dataPoint.as_string())
                # end if
            # end for
        # end if
    # end for
# end while
```

Output:

```
Connected to base station!
Packet Received
Node 38099
Timestamp 2022-01-14 22:33:48.852368156
DATA:
diagnostic_internalTemp : 23.280001
Packet Received
Node 38099
Timestamp 2022-01-14 22:34:18.852170424
DATA:
diagnostic_internalTemp : 23.480000
```

Node address 38099 is one of my nodes. No other address is coming up. 

What is the misunderstanding on my side?

mglord, 2022-01-21:
Apologies, my mistake! Looking into this now.

mglord, 2022-01-21:
Unfortunately there is not currently a way to read the WSDA diagnostic information, including internal temp, directly from the device. The WSDA-2000 does send this information directly to SensorCloud when it is connected - if you wanted you could use the SensorCloud API to read the latest value from the device's internalTemperature channel. More information on the SensorCloud API can be found here: [SensorCloud API](https://github.com/LORD-MicroStrain/SensorCloud)

Sorry for the inconvenience! If you decide to read from SensorCloud and run into any issues let us know!