LORD-MicroStrain/MSCL #205: WSDA-200-USB Communication Error

loustalot, 2021-12-28:
I am trying to use WSDA-200-USB with a raspberry PI.

 I have a very simple code 
   mscl::Connection connection = mscl::Connection::Serial("/dev/ttyUSB0");
    

    mscl::BaseStation mybasestation(connection);
   
    
     mybasestation.readWriteRetries(3);
 

     std::cout<<"Comm Protocol: "<<mybasestation.communicationProtocol()<<std::endl;

  I get an exception
Communication error- code Failed to read the Firmware Version (EEPROM 108) from the BaseStation

   This code works just fine if I plug a WSDA-Base-101 on the USB. 

   Note that the gateway is working fine with SensorConnect. 

   MSCL version is 62.1.2

   Any suggestions?

   Thank you

mglord, 2022-01-05:
I see you've increased the `readWriteRetries` - I definitely recommend leaving this change even though that didn't seem to be the cause of this specific issue.

Another common cause for this error, and what seems to the be problem in this case, is incorrectly specifying the baud rate when instantiating the serial `Connection` object. If not specified the baud rate is assumed to be 921600 (for some WSDAs, like the -101). The baud rate for the WSDA-200-USB is 3000000. If you change the connection line to:
```
mscl::Connection connection = mscl::Connection::Serial("/dev/ttyUSB0", 3000000);
```
hopefully it will resolve the issue.

Let us know if that doesn't work!

loustalot, 2022-01-06:
I tried that but still the same issue. It looks like I am unable to access this gateway.

mglord, 2022-01-06:
If you're able to communicate with the base station via SensorConnect it should work with MSCL as well - SensorConnect uses MSCL for all device communication.

The other variable here is that you're running MSCL on a Linux environment - it would be unusual that the -101 would work without it, but can you try installing/reinstalling the WSDA driver for Linux found here: [WSDA Driver](https://github.com/LORD-MicroStrain/Drivers/releases/tag/linux-cp210x-0.2)

Let us know if that doesn't do the trick and we'll look into additional troubleshooting options. Sorry it's giving you trouble!

loustalot, 2022-01-06:
Thanks I will give a try ASAP. I am using sensor connect under windows, MSCL on windows. I will let you know. Thanks

mglord, 2022-01-06:
Sorry, I didn't realize you were on Windows! On Linux the port name format you're using is correct "/dev/ttyUSB0", but on Windows port names are usually formatted COM# i.e. "COM35".

If you look at the device in SensorConnect you will see the port and baud rate listed under Connection in the device info sidebar:
[image]

Another thing to note is you will need to disconnect the device from SensorConnect (via Disconnect button on the bottom of the device info sidebar or by closing SensorConnect) in order to successfully connect and communicate with it in your MSCL application.

I am surprised you're getting a Communication Failed instead of a Connection Failed error message with that port name though.

You do not need any additional drivers to communicate with the device on Windows.

If you've confirmed your port name and baud rate are correct, and there is no other application with an open connection to the device (SensorConnect) I will need you to send the exact code you're running on Windows to help figure out what's going on.

Sorry for the confusion, thanks!

loustalot, 2022-01-06:
My mistake you got it right in the first place. I am using MSCL on Linux (big typo ....)....

  Thanks for your understanding...

loustalot, 2022-01-10:
It is working now. I updated the cp210x driver following your recommendations and it is working now. I can play with my base....
Thank you very much for your great support.

mglord, 2022-01-11:
Glad to hear you got it working!