LORD-MicroStrain/MSCL #207: Error: Failed to read the Firmware Version (EEPROM 108) from the Base Station.

vinod743374, 2021-12-31:
Error: Failed to read the Firmware Version (EEPROM 108) from the Base Station.
**This is the Error I am getting while running my python code.**

**Procedure we follow :**

Step 1 : We downloaded all the required files to import the MSCL library and also, we use the software called sensor connect which is available in the official site of lord sensors and installed in my office system.

Step 2 : We Provided all the required connections to the devices to turn ON, and make Node 65446 to the Idle mode by long press the power button provided in the node.

Step 3 : Now the base station and the available nodes are visible in the sensor Connect app and we connected; we try to plot some of the parameters from the Node we can plotted in the software itself, it works fine in this scenario.

Step 4 : Now we ran our python code to sync with that sampling data of the Node and printout in our python shell.

Now the real problem comes up,
We are unable to print the data from the Node 65446, We got the below error message in the python shell.

**Error: Failed to read the Firmware Version (EEPROM 108) from the Base Station.**

Here I am attaching the Code and the Error message.

Looking for your response for the solution.

# import the mscl library
import sys
# sys.path.append("../../dependencies/Python")
import mscl

# TODO: change these constants to match your setup
COM_PORT = "COM1"
NODE_ADDRESS = 65446

try:
    # create a SerialConnection with the specified COM Port, default baud rate of 921600
    connection = mscl.Connection.Serial(COM_PORT,9600)

    # create a BaseStation with the connection
    baseStation = mscl.BaseStation(connection)
    
    # create a WirelessNode with the BaseStation we created
    node = mscl.WirelessNode(NODE_ADDRESS, baseStation)

    # create a SyncSamplingNetwork object, giving it the BaseStation that will be the master BaseStation for the network
    network = mscl.SyncSamplingNetwork(baseStation)

    # add a WirelessNode to the network
    #   Note: The Node must already be configured for Sync Sampling before
    #   adding to the network, or else Error_InvalidConfig will be thrown.
    # TODO: Repeat this for all WirelessNodes that you want in the network
    print("Adding node to the network...",)
    network.addNode(node)
    print("Done.")

    # can get information about the network
    print("Network info: ")
    print("Network OK: ", network.ok())
    print("Percent of Bandwidth: ", network.percentBandwidth())
    print("Lossless Enabled: ", network.lossless())

    # apply the network configuration to every node in the network
    print("Applying network configuration",)
    network.applyConfiguration()
    print("Done.")

    # start all the nodes in the network sampling. The master BaseStation's beacon will be enabled with the system time.
    #   Note: if you wish to provide your own start time (not use the system time),
    #   pass a mscl::Timestamp object as a second parameter to this function.
    #   Note: if you do not want to enable a beacon at this time, use the startSampling
    #   noBeacon() function. (A beacon is required for the nodes to actually start sending data).
    print("Starting the network...",)
    network.startSampling()
    print("Done.")

    # =======================================================================================
    # Many other functions are available for the SyncSamplingNetwork:
    #
    # network.lossless()			//enable or disable "lossless" mode for the network (default of enabled).
    # network.ok()					//check whether or not the network is "OK" meaning all nodes fit in the network and have communicated successfully.
    # network.percentBandwidth()	//get the percent of bandwidth for the entire network.
    # network.refresh()				//refreshes the entire network. Should be called any time a change is made to the node after it has been added to the network.
    # network.removeNode()			//remove a node from the network.
    # network.getNodeNetworkInfo()	//get network information for an individual node in the network (TDMA address, percent bandwidth, etc.)
    # =======================================================================================

except Exception as e:
    print ("Error:",e)

mglord, 2022-01-04:
For read/write failures, the first thing I would recommend is to try increasing the `BaseStation::readWriteRetries()` to two or three (ex: `baseStation.readWriteRetries(3)`) any time you instantiate a `BaseStation` object.

I recommend doing the same for any `WirelessNode` objects as well (`node.readWriteRetries(3)`).

Below I've included a snippet from your code with the increased readWriteRetries added.

Do you know which line is throwing this error? Does it always error in the same place?

Let us know if increasing the retries does not resolve the error!

```
try:
# create a SerialConnection with the specified COM Port, default baud rate of 921600
connection = mscl.Connection.Serial(COM_PORT,9600)

# create a BaseStation with the connection
baseStation = mscl.BaseStation(connection)

# create a WirelessNode with the BaseStation we created
node = mscl.WirelessNode(NODE_ADDRESS, baseStation)

# increase readWriteRetries for WirelessNode and BaseStation
baseStation.readWriteRetries(3)
node.readWriteRetries(3)

# create a SyncSamplingNetwork object, giving it the BaseStation that will be the master BaseStation for the network
network = mscl.SyncSamplingNetwork(baseStation)
```

vinod743374, 2022-01-05:
@mglord 
As you recommended I tried by increasing the number of tries, but the error is same,

This is the line throwing the error,

    **network = mscl.SyncSamplingNetwork(baseStation)**
    
Yes, The error is always in the same place.

mglord, 2022-01-05:
Oh! I didn't notice it before - I think the issue may be that you're opening the connection with a baud rate of 9600. I don't think any of our base stations communicate at 9600, but I could be mistaken. The baud rate for the WSDA-200-USB is  3000000 but if you have different model let me know and I can look it up for you. You can also find the information on the device data sheet in the Specifications -> Connectivity section.
```
# create a SerialConnection with the specified COM Port, default baud rate of 921600
connection = mscl.Connection.Serial(COM_PORT,3000000)
```
Sorry for the confusion! Hopefully this resolves the issue - let us know!

vinod743374, 2022-01-10:
@mglord 

thanks for the response but the issue is not resolved,
My Data aggregator model is WSDA - 2000 
When I change the baud rate in the python code to 3000000 , it finds a connection error.

I looked into the datasheet of the WSDA-2000 I found nothing regarding the Baud rate,
I am attaching the screenshots of those, Please look in to it.

Thankyou.
[image]
[image]

vinod743374, 2022-01-10:
@mglord 
For your information,
We are Connecting the WSDA-2000 to the Computer with the Help of an USB cable,
While checking with the sensor connect app it works fine, we can find the data plotting in the data tab.

vinod743374, 2022-01-10:
[image]
[image]

mglord, 2022-01-11:
Thanks for the clarification! I don't know off the top of my head if the WSDA-2000 will communicate over USB the same way our other WSDAs do - I will look into this and get back to you shortly.

vinod743374, 2022-01-12:
@mglord 
I am attaching the Screenshot of the Device info and Properties as well as the data tab.
[image]
[image]

mglord, 2022-01-12:
Sorry for the confusion, thanks for sending the device info!

Unlike our other WSDAs, the WSDA-2000 when connected with USB creates a virtual Ethernet port and communicates via TCP over USB. Once the connection is established all the same base station functions for configuring, initializing a network, and getting data can be used but to establish the connection you'll need to use `mscl.Connection.TcpIp()` instead of `mscl.Connection.Serial()`. With the connection details in your screenshot from SensorConnect, try changing your connection line to:
```
connection = mscl.Connection.TcpIp("fd7a:cafa:0eb7:6578:0001:6105::1",5000)
```
I think I've typed it out correctly, but please double check that I have the address string correct if it doesn't work - this value can be highlighted and copied directly out of SensorConnect. It can also usually be found on the device on the sticker label with the Serial Number, **USB IP**, and Cloud ID.

Hopefully this resolves the issue - let us know how it goes!

vinod743374, 2022-01-19:
@mglord 
Thanks,
It works, Data is printing,
Small doubts regarding the data, As per the python code the data is printing, But the data is printing from the different timestamp, Can you please clarify why it is like that.

[image]

I fetched the data yesterday 6PM, but it is printing from the 12 Noon.
and the Values are also like -64 and 0 like that.

Can you please give us some clarification.

mglord, 2022-01-21:
Regarding the timestamp - the WSDA-2000 has a user-configurable time source. You may just need to manually set the system time, but there are a number of time options available in the WSDA Control Panel (which can be launched from SensorConnect). More details about these options can be found in section **4.3.3 Time Options** of the user manual: [WSDA-2000 User Manual](https://www.microstrain.com/sites/default/files/wsda-2000_manual_8500-0090.pdf)

As far as the values go, it would be helpful to know the node model and which channels are being printed. Are you printing the data with the `WirelessDataPoint::as_string()` function or are you reading it out in a different format? It may be helpful if you include your code where the data is being printed.

vinod743374, 2022-02-17:
# MSCL Example: ParseData
#   This example shows how to parse incoming data from a Wireless Node.
#   This example does not start a Node sampling. To receive data, a Node
#   must be put into a sampling mode (Sync Sampling, Low Duty Cycle, etc.)
#
# Updated: 11/02/2015

# import the mscl library
from __future__ import absolute_import
from __future__ import print_function

import sys
sys.path.append("../../dependencies/Python")
import mscl
#!/usr/bin/python3
import splunklib.client as client
import splunklib.results as results

HOST = "127.0.0.1"
PORT = 8089
USERNAME = "admin"
PASSWORD = "PLmQAz@913750"

# Create a Service instance and log in
service = client.connect(
    host=HOST,
    port=PORT,
    username=USERNAME,
    password=PASSWORD)

try:
    # create a TcpIpConnection with the specified Values printed on the Device
    connection = mscl.Connection.TcpIp("fd7a:cafa:0eb7:6578:0001:6105::1",5000)

    # create a BaseStation with the connection
    baseStation = mscl.BaseStation(connection)

    # endless loop of reading in data
    while True:

        # get all of the data sweeps that have been collected by the BaseStation, with a timeout of 500 milliseconds
        sweeps = baseStation.getData(500)

        for sweep in sweeps:
            # print out information about the sweep
            #print(sweep)
            print ("Packet Received"),
            print ("Node", sweep.nodeAddress())
            print ("Timestamp", sweep.timestamp())
            print ("Sample Rate", sweep.sampleRate().prettyStr())
            print ("Base RSSI: ", sweep.baseRssi())
            print ("Node RSSI: ", sweep.nodeRssi())

            print ("DATA: ")
            # iterate over each point in the sweep
            for dataPoint in sweep.data():
                
                #print out the channel data
                #Note: The as_string() function is being used here for simplicity. 
                #      Other methods (ie. as_float, as_uint16, as_Vector) are also available.
                #      To determine the format that a dataPoint is stored in, use dataPoint.storedAs().
                print (dataPoint.channelName(), ":", dataPoint.as_string())
                searchquery_oneshot = " | makeresults | eval ch="+"\""+dataPoint.channelName()+"\""+",value="+dataPoint.as_string()+"| collect index=sensordata "
                print(searchquery_oneshot)
                oneshotsearch_results = service.jobs.oneshot(searchquery_oneshot)
                
            print ("")
            

except Exception as e:
    print ("Error:", e)

The above code is using to print the data.

mglord, 2022-02-23:
I'm going to go ahead and close this because you've opened the same issue in #215. Please feel free to re-open this if you have additional questions regarding connection issues or data timestamping. Thanks!

closed 2022-02-23 by mglord