LORD-MicroStrain/MSCL #208: EEPROM error while connecting to WSDA-Base-200 (USB)

sgaur24, 2022-01-03:
Hello,

I am trying to connect to basestation WSDA-Base-200 using MSCL python3 library on Ubuntu 18, but keep getting following error:

**Error:**
(base) :~$ sudo python3 BaseStationBeacon.py 
Failed to ping the Base Station
Error: Failed to read the Model Number (EEPROM 46) from the BaseStation
--

The USB stick LED consistent green light, indicating it's working well. Attaching the python script (exactly same as listed on MSCL examples) along with following additional information:

**dmesg output:**
[105986.422082] usb 1-6: new full-speed USB device number 5 using xhci_hcd
[105986.756806] usb 1-6: New USB device found, idVendor=199b, idProduct=ba30
[105986.756808] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[105986.756810] usb 1-6: Product: WSDA-Base-200
[105986.756811] usb 1-6: Manufacturer: LORD Sensing Systems
[105986.756812] usb 1-6: SerialNumber: f4a2282003f7e71195552fb25cc30bca
[105986.759789] cp210x 1-6:1.0: cp210x converter detected
[105986.772897] usb 1-6: cp210x converter now attached to ttyUSB0

**lsusb output:**
Bus 001 Device 007: ID 199b:ba30 MicroStrain, Inc. 

**lsusb-t output:**
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 480M
    |__ Port 2: Dev 7, If 0, Class=Vendor Specific Class, Driver=cp210x, 12M

Thanks,
-sg

[BaseStationBeacon_py.txt](https://github.com/LORD-MicroStrain/MSCL/files/7802629/BaseStationBeacon_py.txt)

mglord, 2022-01-06:
A common cause for this error is incorrectly specifying the baud rate when instantiating the serial Connection object. If not specified the baud rate is assumed to be 921600 (for some WSDAs, like the -101). The baud rate for the WSDA-200-USB is 3000000. Please try updating the connection line to:
```
connection = mscl.Connection.Serial(COM_PORT, 3000000)
```
Also, if you haven't already, you will need to install our WSDA driver for Linux, which can be found [here](https://github.com/LORD-MicroStrain/Drivers#linux), in order to successfully connect to and communicate with the device.

Let us know if neither of these resolves the issue!

mglord, 2022-01-21:
I hope you were able to get this resolved! I'm going to go ahead and close this issue, but feel free to re-open it or another if you continue to have trouble or have any further questions. Thanks!

closed 2022-01-21 by mglord