LORD-MicroStrain/MSCL #210: getDiagnosticInfo

lalandef, 2022-01-06:
I am trying to use getDiagnosticInfo to retrieve the lowBatteryFlag and the current state (sample, idle or sleep) of the TC-200-OEM node connected to a WSDA-101-LXRS analog base station, but I keep getting an error in my Python code.

  Message=Failed to read the Model Number (EEPROM 46) from Node 54521

In SensorConnect, I have the diagnostic info interval On and set to 30 seconds.

What am I doing wrong?  the rest of the code works, retrieving the measured temperature from the node

My Code:

import mscl

#create the connection object with port and baud rate
iCOMport = 3
sCOMport = "COM" + str(iCOMport)
#print (f"sCOMport: {sCOMport}")
connection = mscl.Connection.Serial(sCOMport, 921600)
#create the BaseStation, passing in the connection
basestation = mscl.BaseStation(connection)
# create a WirelessNode with the BaseStation we created
iNodeID = 54521
node = mscl.WirelessNode(iNodeID, basestation)

while True:
    #read data
    sweeps = basestation.getData(500)
    for sweep in sweeps:
        data = sweep.data()           # get the vector of data in the sweep
        for dataPoint in data:
            fTemp=dataPoint.as_float() # get the value as a float
        print("%.3f" % fTemp)
        test= node.getDiagnosticInfo()
        print(test)

mglord, 2022-01-07:
There are two ways to get our wireless nodes to output diagnostic data, with different ways to access the information via MSCL for each:
1. Configure to output a data packet at a regular interval
    - Data packet sent **continuously** at configured interval **regardless of node state (idle, sampling, sleep)**
    - Data packets parsed and stored with the rest of the data received by the base station in MSCL - **they are accessed via `basestation.getData()`**
        - when looping through the returned `DataSweep` objecst, diagnostic data sweeps will have a `samplingType()` of `mscl::DataSweep::SamplingType::samplingType_Diagnostic`
2. Get the diagnostic on-demand with the Get Diagnostic Info command
    - Node sends a **single response** to this command when in **idle mode only**
    - Diagnostic data is sent from the node within a command response - MSCL parses the response and returns the diagnostic data points immediately from the **`node.getDiagnosticInfo()` function**

It sounds like you have your node configured correctly to monitor diagnostic information regardless of node state. I suspect `node.getDiagnosticInfo()` is throwing an error because the node is in sampling mode and cannot respond to any command except Set to Idle. This is fine because the diagnostic data will be accessible from `basestation.getData()`. I recommend also printing out `dataPoint.channelName()` to more easily identify diagnostic data values when viewing the application output for troubleshooting. Some example code for parsing out the diagnostic data from other data can be found on issue #150.

Sorry for the confusion - hope this helps!

lalandef, 2022-01-11:
I got it working, thanks.  node.getDiagnosticInfo() only works in idle like you said, and I got the following Pythn code to read the node state and battery level.  

Thanks for the help

import mscl

# get the version of MSCL (MSCL_VERSION.majorPart(), .minorPart(), and .patchPart() are also available)
print(f"MSCL version: {mscl.MSCL_VERSION}")
#create the connection object with port and baud rate
iCOMport = 3
sCOMport = "COM" + str(iCOMport)
print (f"sCOMport: {sCOMport}")
connection = mscl.Connection.Serial(sCOMport, 921600)
#create the BaseStation, passing in the connection
basestation = mscl.BaseStation(connection)
# create a WirelessNode with the BaseStation we created
iNodeID = 54521
node = mscl.WirelessNode(iNodeID, basestation)

while True:
    #read data
    sweeps = basestation.getData(500)
    for sweep in sweeps:
        data = sweep.data()           # get the vector of data in the sweep
        if sweep.samplingType() == 0: # sampling data
            for DataPoint in data:
                fTemp=DataPoint.as_float() # get the value as a float
            print("%.3f" % fTemp)
        elif sweep.samplingType() == 9:  #diagnostic data
            for DataPoint in data:
                if DataPoint.channelId() == 112 or DataPoint.channelId() == 118:
                    # 112 state: 0 = idle, 1 = sleep, 2 = sampling
                    # 118 battery: 0 = good, 1 = low, 2 = critical
                    print (f"Diag: {DataPoint.channelName()} = {DataPoint.as_string()}")

closed 2022-01-11 by lalandef