LORD-MicroStrain/MSCL #212: Downloading logged sweeps

BaldeepSinghAtMack, 2022-01-25:
How can i accelarate Download time?

mglord, 2022-01-25:
As @ardLORD mentioned in email, there's not a ton of options for increasing download speed. Improving communication with the node by decreasing distance or removing any other potential interference can decrease the need for retries, but a 200 series node will take about 5 minutes to download its full memory capacity.

If your implementation is taking significantly longer than that, I'd be happy to take a look and see if I notice anything in your code that may be slowing it down!

mglord, 2022-02-23:
I hope you were able to find faster solution for data download! I'm going to close this issue, but if you have additional questions feel free to re-open it. Thanks!

closed 2022-02-23 by mglord