LORD-MicroStrain/MSCL #213: Compiling for ARMv7l architechture

cnicho35, 2022-02-02:
Is it possible to compile MSCL for an ARMv7l architecture which exist on an FPGA SoC. The NI compactRIO has linux but it is unable to install debian or redhat packages. The package manager available for this Linux flavor is opkg (https://openwrt.org/docs/guide-user/additional-software/opkg).

I have the toolchain to cross compile for this architecture but I do not know how to build the source for this environment and then wrap it for use in python. Can anyone help me figure out how to do this?

mglord, 2022-02-07:
I don't have any experience with this specific architecture but we have build instructions for Python on Linux: [Building MSCL for Linux](https://github.com/LORD-MicroStrain/MSCL/blob/master/BuildScripts/buildReadme_Linux.md)
These instructions/build scripts are specifically for debian or redhat builds but it seems that would be a good place to start.

mglord, 2022-05-11:
I hope you were able to get this working! I'm going to go ahead and close this, but feel free to re-open if you have any further questions!

closed 2022-05-11 by mglord

cnicho35, 2022-05-11:
Never got it to work - we ended up switching our hardware to an x86 processor which simplified things a lot.