LORD-MicroStrain/MSCL #214: Node Discovery / Each Node Entry duplicated....

BaldeepSinghAtMack, 2022-02-06:
[image]

BaldeepSinghAtMack, 2022-02-06:
Hello,
i am trying to use basestation.NodeDiscovery() to make a List of present nodes. I get each node as Duplicate with a different timestamp. Why?
Please help!

mglord, 2022-02-07:
When a node is powered on it sends out two Node Discovery Packets over each supported frequency - I suspect the duplicate is sent to help compensate for any potential interference. It looks like your base station is receiving both packets for each node, you'll just need to check if the entry already exists.

Sorry for the confusion!

mglord, 2022-02-23:
I hope you were able to get this resolved! I'm going to go ahead and close this issue, but feel free to re-open it if you have any additional questions. Thanks!

closed 2022-02-23 by mglord