LORD-MicroStrain/MSCL #215: How to get the exact acceleration Data (Which is Plotting in the Sensor Connect App) by Using MSCL

vinod743374, 2022-02-21:
Hi 

With the Help of MSCL(Python Code), I am able to print the Data of the Channels CH1 CH2 CH3,
But the Values I am getting is like the below Table,

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/VINODS~1/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/VINODS~1/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

_time | ch1 | ch2 | ch3
-- | -- | -- | --
2022-02-21T10:10:28.000+0530 | 0 | -64 | -64
2022-02-21T10:10:29.000+0530 | -64 | -64 | -64
2022-02-21T10:10:30.000+0530 | -64 | 0 | -64
2022-02-21T10:10:31.000+0530 | -64 | -64 | 0
2022-02-21T10:10:32.000+0530 | -64 | 0 | 0
2022-02-21T10:10:33.000+0530 | 0 | -64 | -64
2022-02-21T10:10:34.000+0530 | 64 | 0 | 0

</body>

</html>

Is there any possibility to get the Values as same as that we are getting in the sensor Connect App.
Please Help Us.

For your Understanding.
We are Using WSDA-2000 Communicating Over TCP/IP
We are using a Wireless Node 64556, 
Model of Node is : G-LINK 200 

Printing the  Data types Using The below Code,

**print (dataPoint.channelName(), ":", dataPoint.as_string())**

 If i change the Data type  as_string() to some other type, Gives Me an error in the Python Code.
 
 
 Thankyou.

mglord, 2022-02-23:
I'm not really sure what might be happening here - SensorConnect uses the same MSCL functions to parse incoming data so there shouldn't be this discrepancy.
The data format from a G-Link-200 can either be a float or integer but both of those should be cast to string (`dataPoint.as_string()`) without issue. If the node is configured to output integer data instead of float the only additional processing SensorConnect does is to convert the data to the correct units, but you would still see the data repeatedly switching between some constant value and zero.

Would you be willing to try the following:
1. Set up and start sampling network via SensorConnect
2. Go to the Data tab and add a Timeseries Graph widget to a dashboard, select the target streaming G-Link channels
3. Collect a few seconds of data, confirm that the data on the graph looks as expected
4. Export a CSV of the collected data from the graph widget (top right corner -> Export Data, the time range will default to the time range currently selected in view in the graph)
5. Close SensorConnect while the node is still sampling
6. Without setting to idle, re-initializing the network, etc, run just the data collection loop of your MSCL application (connect to WSDA port, make `BaseStation` object, jump right into `getData()`) - the code you included in [this comment](https://github.com/LORD-MicroStrain/MSCL/issues/207#issuecomment-1042729087) should do the trick.

Hopefully, since there will be guaranteed no configuration or network changes between collected data on SensorConnect (data values as expected) and MSCL (data value issues) we can get some insight into what may be going wrong. If you could attach the exported .csv from SensorConnect here as well that might be helpful.

Hopefully this helps provide some insight, I'll keep trying to think of what could cause something like this.

vinod743374, 2022-02-24:
**This is how the Data plotting in the Time series graph.**

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/VINODS~1/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/VINODS~1/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

FILE_INFO |   |   |   |   |   |   |   |  
-- | -- | -- | -- | -- | -- | -- | -- | --
ListSeparator= |   |   |   |   |   |   |   |  
DecimalSeparator=. |   |   |   |   |   |   |   |  
  |   |   |   |   |   |   |   |  
SESSION_INFO |   |   |   |   |   |   |   |  
TimeZone | UTC |   |   |   |   |   |   |  
TimeZoneOffset | 0 |   |   |   |   |   |   |  
DateTimeFormat | MM/DD/YY h:mm:ss.nnnnnnnnn |   |   |   |   |  
StartTime | 02/17/22 10:35:25.503906240 |   |   |   |   |   |  
TimeFormat | Standard |   |   |   |   |   |   |  
  |   |   |   |   |   |   |   |  
DEVICE_INFO |   |   |   |   |   |   |   |  
Address | * |   |   |   |   |   |   |  
  |   |   |   |   |   |   |   |  
CHANNEL_INFO |   |   |   |   |   |   |   |  
Channel | Type | SampleRate | Equation | Coefficients | Unit | UnitSymbol | WhereApplied | WhenApplied
65446:ch3 | Timeseries | 256Hz |   |   |   |   |   |  
65446:ch2 | Timeseries | 256Hz |   |   |   |   |   |  
65446:ch1 | Timeseries | 256Hz |   |   |   |   |   |  
  |   |   |   |   |   |   |   |  
DATA_START |   |   |   |   |   |   |   |  
Time | 65446:ch3 | 65446:ch2 | 65446:ch1 |   |   |   |  
02/17/22 10:35:25.503906240 | 0.587649191 | 0.231333 | 0.190775 |   |   |   |   |  
02/17/22 10:35:25.507812490 | 0.576299743 | 0.227602 | 0.179479 |   |   |   |   |  
02/17/22 10:35:25.511718740 | 0.572516593 | 0.217652 | 0.158143 |   |   |   |   |  
02/17/22 10:35:25.515624990 | 0.577560792 | 0.196509 | 0.13053 |   |   |   |   |  
02/17/22 10:35:25.519531240 | 0.59269339 | 0.162928 | 0.096643 |   |   |   |   |  
02/17/22 10:35:25.523437490 | 0.612870186 | 0.118154 | 0.0615 |   |   |   |   |  
02/17/22 10:35:25.527343740 | 0.635569082 | 0.068405 | 0.032633 |   |   |   |   |  
02/17/22 10:35:25.531249990 | 0.657006928 | 0.017412 | 0.012551 |   |   |   |   |  
02/17/22 10:35:25.535156240 | 0.674661624 | -0.02985 | 0.00502 |   |   |   |   |  
02/17/22 10:35:25.539062490 | 0.688533172 | -0.0684 | 0.007531 |   |   |   |   |  
02/17/22 10:35:25.542968740 | 0.693577371 | -0.09577 | 0.015061 |   |   |   |   |  
02/17/22 10:35:25.546874990 | 0.692316321 | -0.11069 | 0.026357 |   |   |   |   |  
02/17/22 10:35:25.550781240 | 0.683488973 | -0.11691 | 0.040163 |   |   |   |   |  
02/17/22 10:35:25.554687490 | 0.669617425 | -0.11567 | 0.05648 |   |   |   |   |  
02/17/22 10:35:25.558593740 | 0.651962729 | -0.1082 | 0.076561 |   |   |   |   |  
02/17/22 10:35:25.562499990 | 0.633046982 | -0.09701 | 0.097898 |   |   |   |   |  
02/17/22 10:35:25.566406240 | 0.614131236 | -0.08209 | 0.116724 |   |   |   |   |  

</body>

</html>

**I tried what you mentioned in the earlier comment, But still the issue is same, I am getting only 0 and -64**

vinod743374, 2022-02-24:
Finally got the solution by changing the data type in the node settings in Sensor Connect app.

Thanks for the Help...

closed 2022-02-24 by vinod743374