LORD-MicroStrain/MSCL #219: Byte Order In Serial Communications Header Wrong?

jkeller-astro, 2022-03-04:
Hello!

I'm building MSCL for x86 under Linux and ran into problems with communicating with serial connected devices.  It SEEMS as if the library implements the byte order handing for the "UE" MIP header incorrectly.  

Specifically, the Start of Packet  is defined as 0x75 followed by 0x65.  The MIP format also states that data is transmitted on the wire in Network Byte Order, and MipPacketParser and MipPacketBuilder treat it as such using read_uint16() and append_uint16() calls respectively.  

That's all fine, but implies that if one were to interpret the first two bytes of a MIP as a uint16, that would be low order byte followed by high order byte :  0x6575.

However,  MIP_PACKET_START_OF_PACKET in is defined as "0x7565" in MipPacket.h.  This causes all communications with a serial microstrain device to not work as the header bytes don't match.  I've attached a patch that uses  MIP_PACKET_SOP1 and MIP_PACKET_SOP2 to be explicit that "0x75" follows "0x65" on the wire, but I don't understand why this isn't broken for everyone!

What am I missing?

-------------------------------------------------------------------------------

diff -ur MSCL-63.1.0.orig/MSCL/source/mscl/Endianness.h MSCL-63.1.0/MSCL/source/mscl/Endianness.h
--- MSCL-63.1.0.orig/MSCL/source/mscl/Endianness.h	2022-01-14 14:06:06.000000000 -0800
+++ MSCL-63.1.0/MSCL/source/mscl/Endianness.h	2022-01-21 17:03:02.838426284 -0800
@@ -42,7 +42,7 @@
 
 #pragma once
 
-#include <boost/detail/endian.hpp>
+#include <boost/endian.hpp>
 #include <algorithm>
 
 namespace mscl
@@ -118,7 +118,7 @@
 inline StaticType SystemEndian_To_LittleEndian(StaticType val);
 
 
-#ifdef BOOST_LITTLE_ENDIAN
+#if __BYTE_ORDER == __LITTLE_ENDIAN
 // Little endian system
 
 template<typename StaticType>
diff -ur MSCL-63.1.0.orig/MSCL/source/mscl/MicroStrain/MIP/MipParser.cpp MSCL-63.1.0/MSCL/source/mscl/MicroStrain/MIP/MipParser.cpp
--- MSCL-63.1.0.orig/MSCL/source/mscl/MicroStrain/MIP/MipParser.cpp	2022-01-14 14:06:06.000000000 -0800
+++ MSCL-63.1.0/MSCL/source/mscl/MicroStrain/MIP/MipParser.cpp	2022-01-21 17:01:10.921851387 -0800
@@ -313,10 +313,13 @@
         }
 
         //read the first 2 bytes
-        uint16 startOfPacket = data.read_uint16();                //Start of Packet
+        uint8 startOfPacket[2] = {0};
+        startOfPacket[0] =  data.read_uint8();
+        startOfPacket[1] =  data.read_uint8();  
 
         //verify that the Start of Packet value is correct
-        if(startOfPacket != MipPacketInfo::MIP_PACKET_START_OF_PACKET)
+        if(startOfPacket[0] != MipPacketInfo::MIP_PACKET_SOP1 ||
+           startOfPacket[1] != MipPacketInfo::MIP_PACKET_SOP2)
         {
             //Invalid Packet
             return mipParserResult_invalidPacket;
@@ -375,7 +378,8 @@
 
         //build the checksum to calculate from all the bytes
         ChecksumBuilder calcChecksum;
-        calcChecksum.append_uint16(startOfPacket);
+        calcChecksum.append_uint8(startOfPacket[0]);
+        calcChecksum.append_uint8(startOfPacket[1]);
         calcChecksum.append_uint8(descriptorSet);
         calcChecksum.append_uint8(payloadLen);
         calcChecksum.appendBytes(payload);
diff -ur MSCL-63.1.0.orig/MSCL/source/stdafx.h MSCL-63.1.0/MSCL/source/stdafx.h
--- MSCL-63.1.0.orig/MSCL/source/stdafx.h	2022-01-14 14:06:06.000000000 -0800
+++ MSCL-63.1.0/MSCL/source/stdafx.h	2022-01-21 17:03:00.950416477 -0800
@@ -15,7 +15,7 @@
 #include <boost/date_time.hpp>                                     //for boost::posix_time::from_time_t
 #include <boost/date_time/posix_time/posix_time_duration.hpp>      //for boost::posix_time::nanosec
 #include <boost/date_time/posix_time/ptime.hpp>                    //for boost ptime    
-#include <boost/detail/endian.hpp>                                 //for endianess
+#include <boost/endian.hpp>                                        //for endianess
 #include <boost/numeric/conversion/cast.hpp>                       //for boost::numeric_cast
 #include <boost/optional.hpp>                                      //for boost::optional
 #include <boost/utility/binary.hpp>                                //for BOOST_BINARY

karljohanreite, 2022-05-13:
Please see https://github.com/sintef-ocean/conan-mscl/commit/3f96a24181f86ea4dd37150900119ef0c66a13f0

jkeller-astro, 2022-06-03:
> Please see [sintef-ocean/conan-mscl@3f96a24](https://github.com/sintef-ocean/conan-mscl/commit/3f96a24181f86ea4dd37150900119ef0c66a13f0)

Interesting.  I'll give that a try.