LORD-MicroStrain/MSCL #236: Is it possible to read data into Excel in real time, if so how? I'm not sure if I can link through SensorConnect or if I would have to use MSCL.

bcicerone, 2022-05-05:


mglord, 2022-05-11:
It may be possible, but we do not offer any Excel-specific support.

For custom/application-specific data handling I would definitely recommend using MSCL. If you have any questions about getting MSCL installed, establishing communication with your device, parsing data, etc. we'd be happy to help!

There isn't really a great way to get data output from SensorConnect in real time. You can export CSVs through the Time Series Graph widget but that doesn't sound like it will meet your needs. I guess you could read from the raw data files as they're being written but I can't say I'd recommend that.

salmentoj, 2023-04-21:
Msclissa, Can you let me know where to find the format for the raw data files? I would like to read the data directly into excel using a VBA macro while SensorConnect is running. The macro will only be collecting data every 5 to 30 seconds.

closed 2024-10-24 by github-actions[bot]