LORD-MicroStrain/MSCL #237: Updated Boost version

karljohanreite, 2022-05-11:
When building MSCL with Boost version 1.79, final executables are not able to connect to an inertial node available via an tcpip server.  Both ping() and interrogation of e..g. node name times out. 

It would be nice to support newer versions of Boost.

mglord, 2022-05-11:
Unfortunately we have no immediate plans to update the Boost dependency version. It's certainly something we should look into though.

In general, communication with inertial devices over TCP/IP should work fine. If you run into any issues establishing successful communication with Boost 1.68 or SensorConnect let us know and we can help determine what may be going wrong.

Sorry for the inconvenience, but thanks for reaching out!

karljohanreite, 2022-05-12:
Thank you for your swift answer! 

The background is that we are making conan packaging for MSCL.. See https://github.com/sintef-ocean/conan-mscl. We use this as one of many dependencies in our software (https://ratatosk.smd.sintef.no) which communicates with a Microstrain MRU, among other things. All other dependencies are building with Boost 1.78. 

Do you have any thoughts about what could cause timeout in tcp-ip connections when building with newer Boost? I see warnings about the endianness library and we also will look into how boost asio is used.

karljohanreite, 2022-05-13:
I think I found a way to use newer Boost version. At least it seems to solve my issues with a Microstrain MRU. The fix in our conanfile is to perform the following replacements:

```
for f in ["MSCL/source/stdafx.h", "MSCL/source/mscl/Endianness.h", "MSCL_Unit_Tests/Test_Utils.cpp"]:
	tools.replace_in_file(
		os.path.join( str(self.source_folder), "MSCL", f),
		"boost/detail/endian.hpp",
		"boost/predef/other/endian.h")

for f in ["MSCL/source/mscl/Endianness.h", "MSCL_Unit_Tests/Test_Utils.cpp"]:
	tools.replace_in_file(
		os.path.join( str(self.source_folder), "MSCL", f),
		"BOOST_LITTLE_ENDIAN",
		"BOOST_ENDIAN_LITTLE_BYTE")
```

I think this fix should be applicable for all newer Boost versions, and I suggest that they are considered for MSCL . Also, if you consider making MSCL available through conan, please see https://github.com/sintef-ocean/conan-mscl.

closed 2024-10-24 by github-actions[bot]