LORD-MicroStrain/MSCL #244: setInitialFilterConfiguration() not supported : how to achieve the same effect ?

EliseLgd, 2022-05-20:
Hello,

I am using a 3DMCX5-AR. When initializing the estimation filter, I wish to let the system calculate the initial pitch and roll with the accelerometers, and to manually set the yaw.

In other words, I want to replicate using the MSCL in C++ the following command in SensorConnect :

[image]

After searching through the library documentation, I wrote the following set of instructions : 
`mscl::FilterInitializationValues initValues = mscl::FilterInitializationValues();
initValues.initialValuesSource = mscl::FilterInitialValuesSource::UserSpecified_Heading;
node.setInitialFilterConfiguration(initValues);`

Unfortunately, the last command does not seem to be supported by the device.
How can I achieve the effect I need ?

msclissa, 2022-05-20:
Sorry for the confusion!

The `setInitialFilterConfiguration()` command at this time is only supported on the GQ7 and combines the functionality of multiple other commands.

The  command you're looking for is `InertialNode::setInitialHeading(float heading)`. Using this command automatically lets the device know to use accelerometers to determine pitch and roll.
If you did want to specify the pitch and roll as well you would use `InertialNode::setInitialAttitude(const EulerAngles& attitude)`

One important thing to note about both of these is the heading/attitude value specified will only take effect if the command is sent while the filter is in Initialization state (Filter State = 0x01). To make sure the filter is in the Initialization state, issue a filter reset command (`InertialNode::resetFilter()`) prior to configuring. Receiving the initial heading/attitude in Initialization state should transition the filter to the Running state.

Hope this helps!

EliseLgd, 2022-05-23:
Thank you for your answer, this helps indeed!

However, it seems that when using this command, the filter takes approximately 5 seconds to start issuing valid values again. I have been performing several tests and always get this time interval during which the filtered attitude is (0, 0, 0). I did not have this problem when using `InertialNode::setInitialAttitude(const EulerAngles& attitude)` but I get it with `InertialNode::setInitialHeading(float heading)`. Is it avoidable ?

closed 2024-10-24 by github-actions[bot]