LORD-MicroStrain/MSCL #261: uint16 return values for offset and slope values

saytesla, 2022-06-27:
We are currently working on a project in python which is using the MSCL Library. We are extracting  acceleration data(x,y,z) from a vibration sensor. We were wondering if there was a way to extract uint16 values for the slope and offset values. Currently, the functions provided in MSCL return floating point values for the slope and offset values. 

The code below is an example of how we are getting values for the offset and slope.
Chgroups = node.features().channelGroups()
Offset = node.getLinearEquation(chgroups[0].channels()).offset())

msclissa, 2022-06-27:
Would you mind clarifying what you're asking? The slope and offset are configured and stored on the device as floats - there is no way to interpret them as uint16 values without losing precision.

Many of our nodes have the option to stream uncalibrated, truncated int16 data (20 bit integer shifted right to 16 bits) - when MSCL processes this data it is automatically bit-shifted back to where it should be so the data point value can be accessed (`as_int32()` because it needs the additional space after being shifted) and the slope and offset applied without any further steps needed to interpret the data point value properly.

For any uncalibrated streamed data output from our wireless sensors, MSCL interprets it so all that needs to be done to get the actual calibrated value is multiply it by the slope and add the offset.

I hope this helps! If not, it may be helpful to know why you're looking for the slope and offset values as uint16s specifically.

saytesla, 2022-07-05:
Hi, 
How can I get to stream int16 data? For bandwith issues, we are looking to stream int16 data instead of floating point numbers.

msclissa, 2022-09-08:
You can configure the node to stream int16 data via the `WirelessNodeConfig` object:
```
mscl::WirelessNodeConfig nodeConfig;
nodeConfig.dataFormat(mscl::WirelessTypes::DataFormat::dataFormat_raw_int16);
node.applyConfig(nodeConfig);
```
As outlined above, though, when you read the data out of the `WirelessDataPoint` object, you should access it via `dataPoint.as_int32()`. The data is transmitted and received as the truncated int16 value but MSCL the shifts it back to a meaningful value.

The slope and offset are stored and retrieved as float values - unless they change they only need to be read once. To calibrate the data, multiple the datapoint value (`as_int32()`) by the slope and add the offset - this will result in the calibrated floating point value.

closed 2023-04-27 by saytesla